yaqwsx/KiKit #120: [bug] vtabs/htabs swap meaning between grid and tightgrid

gwideman, 2021-02-03:
In panelize grid, htabs specifies how many tabs kikit places between vertical edges. Ie: htab = the tabs span horizontally between boards.

In panelize tightgrid, you have to instead use vtabs to spec how many tabs between vertical edges.

yaqwsx, 2021-02-03:
I confirm this is a bug. However, fixing it is a breaking change for existing projects; thus I suggest fixing this when we update the CLI (as it will be already a breaking change).

yaqwsx, 2021-03-23:
This is fixed in #139 as there is no longer a difference between grid and tight grid. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

I suggest moving the discussion to the PR.

closed 2021-03-23 by yaqwsx