yaqwsx/KiKit #121: Missing framecuts for tightgrid?

gwideman, 2021-02-03:
'panelize grid' has --framecutH/V, but that seems to be missing from 'panelize tightgrid'.

That's probably not an issue when using V-cuts, since they will cut the frame too.

However, when using mousebites, there seems to be no way to cut the frame.  Am I missing something?

yaqwsx, 2021-02-03:
The tightgrid is a historical strategy implemented in time when KiKit did not support frames nor frame cuts. It was designed primary for V-cuts separed boards with irregular shape. Most of the fabhouses out there mill the V-cuts manually/semi-automatically, thus they require a straight edges of the panel (they use is as a reference edge for the slot milling). Therefore, tightgrid was the easiest shape to implement to have all these features. Since it was primary used for V-cuts, no frame cuts were required.

If you use V-cuts I see no point in using tightgrid. Could you describe your use case?

gwideman, 2021-02-03:
We don't use V-cuts, we use mousebites, because we get better mechanical fit with milled edges. For most purposes 'panelize grid' would work fine.  Unfortunately we have some long narrow boards which we want to arrange in 1 x n (example: gridsize 1 4).  And for various reasons, we can't put the mousebites on the narrow ends of the boards, they have to go on the sides. To get robust support, we need rails _between_ the boards, which I believe kikit can only produce using tightgrid, right?

Here's an example pic: 
[image]

... which leads to the desire for framecut. But it would be just as good if the "grid" command supported interior rails.

This software is so productive, that of course it prompts requests to be even more complete!

jdaandersj, 2021-02-04:
I did not realise that interior rails were possible but I also like this feature

yaqwsx, 2021-02-04:
Yeah, as I mentioned in #119 backbones are on the roadmap.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx