yaqwsx/KiKit #123: No holes in the paste mask layer when using --tooling.

hdeveer, 2021-02-05:
I tested this option and saw in the gerber viewer that there are no alignment holes for the tooling holes in the paste mask, only the component pads. Am I missing something or is it simply not present in the script?

What I would expect is that the paste mask layer would have holes with the exact the same dimensions as the tooling holes specified.
This makes alignment of the stencil much more simple in a manual rig (take one PCB panel, drill holes true the tool holes in a, wooden, board and press alignment pins in there).

These are the most related parameters I used:

kikit panelize grid ..... --tabwidth 3 --tabheight 3  --radius 1.5 --railsTb 5 .. --fiducials 10 2.5 1 2 --tooling 5 2.5 2 ...

Is there a way to workaround this and add the drill-holes in an easy way manually? Or maybe change the pad stack used for the tooling holes to include a hole in the mask?

yaqwsx, 2021-02-05:
The tooling holes in the paster layer are omitted by design as they are ordinary NPTH holes. I am not sure, however, if they should be included or not in the paste layer.

I have never heard that they would be used for stencil alignment in production. If you align them with pins, the stencil will be hard to remove without distorting the applied paste. Usually, the stencil is aligned in a different way.

I need to hear other opinions/use-cases/experiences to decide to change this. 

Currently, you can make your panel and manually change the properties of the tooling holes. If we rework the CLI soon, we can consider adding an option to select if the user wants the tooling on the paster layer or not.

yaqwsx, 2021-02-05:
BTW: A local, dirty solution for you might be changing (https://github.com/yaqwsx/KiKit/blob/master/kikit/resources/kikit.pretty/NPTH.kicad_mod) to be included in the paster layer; however, then all your mouse bites will also be included in the stencil.

jdaandersj, 2021-02-05:
I vote for adding an option for this in new CLI. I can see it could be useful but also that it should not be compulsory.

henri62, 2021-02-05:
Well I modified the 3 pads in kicad project so they become open. So the suggestion you gave in the first reply.

My usecase is the following:
I want to use the holes for manual production, just make a wooden plate, place the board on it and make 2 drill holes (true the holes of the rails exactly vertical). Then press-in 2 pins (2 mm). I can place the board and the stencil on it now and apply the paste.
Remove the stencil, place the components, remove the PCB and put it in an SMT oven.

Ofcourse removing the stencil shall be done with great care to avoid bending it. Without the holes I have no good means to align it.

I also read the discussion about a new way of specifying options for the CLI, so I fully agree with jou and jdaandersj to add an option to switch on a feature that does this for the new CLI design.

aa956, 2021-03-20:
NPTH's in paste mask is certainly unusual requirement.

> I need to hear other opinions/use-cases/experiences to decide to change this.

One of the options may be to allow inclusion of arbitrary footprint at arbitrary location, I imagine the following use cases:

- Text/graphic footprint(s) with part number of the PCB / project name / barcode / logo / visible copper layer number / silkscreen filled rectangle for laser marking in silkscreen/copper/soldermask that does not fit onto single PCB but is useful on the panel rails / free area
- Layer stack-up drawing as footprint
- Some other drawings that are good to have on the panel gerber, e.g. arrow(s) pointing to marking locations (where PCB manufacturer is to place own logo, E-test pass mark, UL marking to not interfere with PCB design etc)
- Custom mounting holes / routes (including this request, custom NPTH footprint with paste mask opening for mounting hole)
- Non-round fiducial marks (although AFAIK these are usually not recommended by SMT equipment manufacturers but there may be some use cases e.g. legacy equipment?)
- Layer alignment targets (are there PCB manufacturers still using them?)
- Anything that is not going onto single PCB but may be useful in the panel - some in-panel test connectors maybe?

henri62, 2021-03-20:
Its a funny coincidence that today I received my PCBs and stencil, and they look good with my manual editing of the feducials into npth holes (on the rails only of course). Now I still need to order parts and assemble them. (And make a picture how I did the alignment.)

yaqwsx, 2021-03-23:
This is implemented in #139. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx