yaqwsx/KiKit #124: Overview of KiKit structure to help us install/use it better?

gwideman, 2021-02-07:
For all that KiKit does some very productive stuff for panelization, the necessity under Windows to run it either in WSL or Docker is painful. I've put a brief note about the pain at the end of this post.

So what I'd really appreciate is a better understanding of what kind of a thing KiKit is.

I already know it's something that can be run from the OS command line, to run the panelization function.  From various pages of docs, I gather it may also add features to the KiCad UI (plug-ins?), though I can't test that as it doesn't install into the Windows hosted instance of KiCad.  And the descriptions in the current KiKit docs don't really explain what form each functionality takes.

Given I just want to run the command-line panelization features, it makes me wonder why I can't just run that Python code under Windows command line (or PowerShell).  

OK, from (https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md) I get the impression that the current installation has some binary dependencies. What are those? And are they needed for just using the command line? And can I just install them manually?  

On the FAQ page, I see the comment "the pcbnew module is not installed for your system Python installation. The KiCAD's Python does not allow to install libraries with binary dependencies, therefore you cannot install KiKit in it."  But my naive model is that panelization runs from the command line outside of KiCad, so would not need binaries installed in KiCad's embedded Python environment.

So it would be a great help if there could be a description of which KiKit functionalities are implemented in which form (external script, or plug in, or script to run in KiCad console, etc), and what dependencies they have. And in particular, what the obstacles there are to at least getting the very useful panelization feature installed and working directly from Windows command line.

Thanks for your efforts!

**The Docker pain...**

After installing Docker (which required installing WSL and enabling Hyper-V) my Windows 10 Pro machine experienced occasional random Blue Screen crashes with DPC Watchdog Violation error. This is after years of running with zero blue screens -- indeed I've never before even seen a Win 10 blue screen.  These occur with no apparent relationship to actually doing anything in Docker or WSL at the time.  There are a bunch of posts on this, without a resolution that I could find. Example: 
https://www.reddit.com/r/techsupport/comments/edbogl/tried_to_run_docker_on_windows_and_got_dpc/.  And it might be WSL.Hyper-V that's the issue rather than Docker per se, except that Docker depends on WSL. But regardless all of that is a big rabbit hole just to "run some python scripts".

yaqwsx, 2021-02-07:
TLDR: On native Windows, nothing works*. I am sorry, however, until KiCAD 6 is released, there is nothing I can do about it. Docker is a nasty workaround, but probably better than nothing. I am sorry to hear about your problems with running Docker, unfortunately, that is out of my control.

* Possibly the modify and export commands work.

More elaborate answer:

KiKit is a standalone tool, however, it uses Pcbnew Python API. This is why it cannot run without KiCAD installed. KiCAD on Windows is compiled using MSYS2. Standard Python interpreter you can download for Windows is compiled using MSVC. You cannot mix libraries compiled by MSYS2 and MSVC (as they use different mangling and probably calling convention - I am not sure about the second one). Therefore, KiCAD on Windows is shipped with a copy of the Python interpreter built under MSYS2. Python packages might use some native libraries. When you install them via pip, their binary form is downloaded from PyPi (those libraries are compiled using MSVC, so they work fine with the interpreter installed from python.org). However, you cannot use native libraries from PyPi in the Python interpreter shipped with KiCAD (as it is built using MSYS2). The same applies the vice-versa - you cannot load the Pcbnew library to the interpreter built under MSVC (the one downloaded from python.org).

KiKit internally uses several native libraries - the essential one is Shapely. However, on Windows, there are two separate worlds - one provided by the KiCAD's Python interpreter, where you can load Pcbnew library, but you cannot load Shapely. The second world is provided by the user-downloaded Python interpreter, where you can load Shapely, but you cannot load the Pcbnew library.

The solution I plan is to have an inter-process procedure call bridge between these two interpreters. Basically, KiKit will run in the python.org interpreter, and when it needs to interact with KiCAD, it will send commands to the KiCAD's Python interpreter. However, this cannot be currently done, as KiCAD is shipped with Python 2 (!!!) on Windows. That makes a no go for any IPC bridge. That's why we have to wait for KiCAD 6 release (note that the current nightly is still shipped with Python 2).

A few notes:
- There are good reasons KiKit has these dependencies - the original (unpublished) version of KiKit was extremely slow as Python is good glue, but offers terrible performance. Also, it was buggy and much longer code, thus harder to maintain. This is why we use Shapely - it is well tested, high-performance library. It allows us to focus more on the KiKit specific stuff instead of implementing standard polygon manipulation algorithms.
- I understand why KiCAD decided to use MSYS2 - KiCAD has always been developed for Linux/GNU/Unix and it would be too much work to make it compilable by MSVC. Hence, they use MSYS2. If I was in their position, I would probably do the same thing.
- If KiCAD would be shipped with Python 3 on Windows, it might make sense to go through the trouble, and compile the required native libraries under MSYS2 and distribute them from our channels. However, that would be a lot of extra work to maintain I am not sure I can fit inside my time schedule and since we have no Python 3 support on Windows yet there's no point in even trying.
- KiKit will never support archaic Python 2. The developers are encouraged to switch to Python 3 since 2008 and Python 2 was originally intended to be deprecated in 2015 (but in reality, it was deprecated on 1st 2020).

gwideman, 2021-02-08:
Thanks @yaqwsx for that detailed discussion.   I'd like to be sure I understand the heart of the matter, both as it relates to KiKit, and also other code I might want to run with KiCad.

I'm assuming that there's no problem to run plain old Python 2.7 code using KiCad's Python within MSYS2+MinGW, and such code could call the KiCad API, right?

And presumably there's no trouble in such a program to import libraries that contain only python code.

The trouble arises when importing Python libraries with a binary component, as that binary component would presumably have been compiled with MSVC, and not be compatible with KiCad's Python.

OK, but are such libraries either available compiled with gcc for Windows and MSYS2+MinGW, or are they available as C source code that could be compiled on the users machine with MinGW gcc?

I'm not saying that would be a winning business model :-), just verifying that I have the concept right.

For what it's worth, I found this issue report: "install shapely on msys2 #633"  https://github.com/Toblerity/Shapely/issues/633  which seems to indicate that the at least victorliu got Shapely installed on msys2.  (See also https://github.com/msys2/MINGW-packages/issues/6017). 

And I do realize that KiKit uses other libraries with binaries. I guess numpy is available as a compiled library for MSYS2 here https://packages.msys2.org/package/mingw-w64-x86_64-python-numpy?repo=mingw64, but that's for Python 3. So one would need numpy 1.16.6 (last for Python 2.7), and I didn't find that precompiled in a quick look.  

So many dependencies, not enough time....

yaqwsx, 2021-02-08:
> I'm assuming that there's no problem to run plain old Python 2.7 code using KiCad's Python within MSYS2+MinGW, and such code could call the KiCad API, right?
> And presumably there's no trouble in such a program to import libraries that contain only python code.

That is correct.

> The trouble arises when importing Python libraries with a binary component, as that binary component would presumably have been compiled with MSVC, and not be compatible with KiCad's Python.

That is correct. And also for KiKit, that KiCAD on Windows provides Python 2.

 
> For what it's worth, I found this issue report: "install shapely on msys2 #633" [Toblerity/Shapely#633](https://github.com/Toblerity/Shapely/issues/633) which seems to indicate that the at least victorliu got Shapely installed on msys2. (See also [msys2/MINGW-packages#6017](https://github.com/msys2/MINGW-packages/issues/6017)).

Yeah, the procedure is nasty - you have to manually move and rename libraries. This is way too much for basic users. Also, the solution is fragile, so I don't want to support it "officially" in the installation guide.
 
> So many dependencies, not enough time....

Yup, I couldn't agree more. So I hope you now understand my recommendation to use either WSL or a docker image on Windows (at least until we finally get Python 3 on Windows)

gwideman, 2021-02-08:
Again thanks @yaqwsx  for your quick answer, and confirming my understanding.

yaqwsx, 2021-03-18:
There are some positives news regarding the packaging of KiCAD 6 - mentioned in #21. To make things easier to track, I suggest continuing this discussion in #21.

closed 2021-03-18 by yaqwsx