yaqwsx/KiKit #127: Feature request : asymetrical / more fiducials

JulienDevillers, 2021-02-22:
Hello,

Current version builds 3 fiducials per side. All are placed on a symetrical position.
However, some manufacturers request for only 2 fiducials per side at specific position  regarding
the corners and the side (top, bottom) of the board. I suspect this is for side / orientation recognition.

So the request is to be able to choose a relative position regarding the sides and the position of the fiducial. 
I would add another argument named like "singleFiducials" that would be an array of hOffset, vOffset, copperDiameter, openingDiameter and boardSide.

Joined is an example of request.
[panel.pdf](https://github.com/yaqwsx/KiKit/files/6020776/panel.pdf)

Thanks
Julien

yaqwsx, 2021-02-22:
I started to work on CLI rework, which will allow us to specify much more complicated parameters for the panels easily. It should be done soon. Then it will be easy to implement your request.

JulienDevillers, 2021-02-22:
Hum, btw my implémentation Is nearly finished, I need some answers from the
board manufacturer to close it.

I didn't want to do it at first but it came quickly obvious that it would
be faster to implement it rather than DIYing a custom python software
starting from your code....

yaqwsx, 2021-02-22:
I will not accept new options to the old CLI as it is a mess and hard to maintain. You can read about the new CLI proposal in the discussions: https://github.com/yaqwsx/KiKit/discussions/111

Adding a ton of CLI switches is not the way to go. It makes the software only harder to use.

JulienDevillers, 2021-02-22:
I agree.
I added this line to FR : "Shape of the fiducials is usefull too as (se below), some requests, square ones...."

yaqwsx, 2021-03-23:
This is fixed in #139. If you are interested in testing the new CLI, please follow https://github.com/yaqwsx/KiKit/pull/139#issuecomment-804857951.

yaqwsx, 2021-03-23:
This is partially addressed in #139 by:
- fiducial style. Currently, only a simple circular one is implemented, however, it is straightforward to implement more styles (without messy UI).
- the possibility to run a custom post-processing script.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved.

closed 2021-04-19 by yaqwsx