yaqwsx/KiKit #129: Mouse bites appear incorrect

sslupsky, 2021-02-27:
I have been experimenting with KiKit and came across a couple issues.

First, in the screenshot below, the mouse bites do not look correct to my eye:

[image]

Here is how I invoked KiKit to generate the panel:
```
kikit panelize grid --gridsize 2 2 --space 3 \
--tabwidth 5 --tabheight 2 \
--tabsfrom Eco1.User 5 --vtabs 0 \
--mousebites 0.5 1 0.25 \
--radius 1 \
--panelsize 100 100 \
--fiducials 10 2.5 1 2 --tooling 5 2.5 1.5  \
sensor\ interface.kicad_pcb panel.kicad_pcb
```

Second, I would prefer to use vcuts instead of mouse bites but I am observing an error message that indicates you cannot vcut a curve.  
```
An error occurred: Cannot V-Cut a curve
```

In the screenshot below, I think KiKit is complaining about the curve on the left hand side of the board?  KiKit has placed a tab here.  Can we override the curve and tell KiKit to place the vcut to cut the horizontal tab?

[image]

Original board showing the tab vectors:
[image]

sslupsky, 2021-02-27:
Setting htabs 0 helped fix the irregular mouse bits on the lower instances

[image]

But the two instances along the top have mouse bites all around the perimeter still.

sslupsky, 2021-02-27:
Odd, I deleted the top tab vector and drew it again.  This time the mouse bites look correct:

[image]

sslupsky, 2021-02-27:
So, I think the only "question" I have is with regard to the V-Cuts.  There top, bottom and right side have straight line segments.  The left side is a curve.  It would be handy if I could specify a V-Cut for that curve and create a "flat spot" without having to add a flat segment to the board outline.

sslupsky, 2021-02-27:
Or optionally, have a combination of mouse bites and V-cuts.  So, the straight segments would be v-cuts and the curve would be mouse bites.

sslupsky, 2021-02-28:
I think I may have encountered another mouse bite issue (I haven't been able to find the correct combination of parameters):
```
kikit panelize \
tightgrid \
--slotwidth 2.5 \
--panelsize 300 200 \
--gridsize 3 6 \
--space 10 \
--tabsfrom Eco1.User 5 \
--tabsfrom Eco2.User 3 \
--vtabs 0 \
--htabs 0 \
--mousebites 0.5 1 0.25 \
--radius 1 \
--fiducials 10 2.5 1 2 --tooling 5 2.5 1.5  \
sensor\ interface.kicad_pcb panel.kicad_pcb
```

[image]

yaqwsx, 2021-02-28:
Helo, the behavior you observe is really strange and definitely should not be happening. Could you share the board with me, so I can reproduce and investigate the issue?

About V-cuts: V-Cuts are straight lines. So KiKit will refuse to perform V-cuts if the cut is curved unless you specify `--vcutcurves` - in that case, the cut is approximated by a line going through the first and the last point of the cut.

sslupsky, 2021-02-28:
Here you go.
[kikit_issue_129.kicad_pcb.zip](https://github.com/yaqwsx/KiKit/files/6057225/kikit_issue_129.kicad_pcb.zip)

Here are the parameters I use to recreate this:
```
kikit panelize \
tightgrid \
--slotwidth 2.5 \
--panelsize 300 200 \
--gridsize 3 6 \
--space 10 \
--tabsfrom Eco1.User 5 \
--tabsfrom Eco2.User 5 \
--vtabs 0 \
--htabs 0 \
--mousebites 0.5 1 0.25 \
--radius 1 \
--fiducials 10 2.5 1 2 --tooling 5 2.5 1.5  \
kikit_issue_129.kicad_pcb panel.kicad_pcb
```

sslupsky, 2021-02-28:
The purpose of the tab vectors is to locate the tabs along the straight line segments between the arcs.

sslupsky, 2021-02-28:
Another peculiar issue I have noticed with this particular board outline is if I increase the length of the tab vector outside the board outline from 2.54mm to 5.08mm, the tabs disappear.

[image]

sslupsky, 2021-02-28:
Here is another configuration of parameters that generates an odd output:
```
kikit panelize \
grid \
--panelsize 180 85 \
--gridsize 2 5 \
--space 5.0 \
--radius 1 \
--mousebites 0.5 1 0.25 \
--fiducials 10 2.5 1 2 \
--tooling 5 2.5 1.5  \
sensor\ interface.kicad_pcb sensor\ interface\ panel.kicad_pcb
```

[image]

yaqwsx, 2021-04-13:
During the implementation of #139 I might have found reason for the failure you describe. However, the cut algorithm have been revwritten in #139, so it should no longer suffer from this problem.

yaqwsx, 2021-04-19:
#139 was merged to master. Therefore this issue should be resolved. Please check it.

yaqwsx, 2021-04-26:
There was no activity for the last week, closing. If the same problem is preserved, feel free to reopen.

closed 2021-04-26 by yaqwsx

bradanlane, 2021-05-14:
I was having many of the same issues described here panelizing a round PCB. installed latest `master` _and after some mistakes with the new JSON syntax_ created the exact test panel I needed .THANKS (and enjoy the kofee)

[image]