yaqwsx/KiKit #132: Fabrication JLCPCB_IGNORE field fails to ignore the component

sslupsky, 2021-02-28:
I created a field in a schematic with the name "JLCPCB_IGNORE".  I filled the field with "do not fit" for the specific components I want to ignore.  I observe that these components remain in the BOM and POS output files generated by the fabrication feature.

In the screenshots below, J1 R1 R5 and RV1 have the JLCPCB_IGNORE field set to "do not fit" but remain in the BOM file. 
```
kikit fab jlcpcb --assembly --field LCSC\ PN --schematic sensor\ interface.sch sensor\ interface.kicad_pcb fabrication/jlcpcb
```
[image]

[image]

sslupsky, 2021-02-28:
I have confirmed the field is defined in the schematic file:
```
F 5 "do not fit" H 3050 3850 50  0001 C CNN "JLCPCB_IGNORE"
...
F 10 "do not fit" H 7750 3550 50  0001 C CNN "JLCPCB_IGNORE"
...
F 9 "do not fit" H 4350 3600 50  0001 C CNN "JLCPCB_IGNORE"
...
F 9 "do not fit" H 4350 3600 50  0001 C CNN "JLCPCB_IGNORE"
```

With regard to the POS file, it doesn't appear that the components are filtered here:
https://github.com/yaqwsx/KiKit/blob/55484ff6dcc6114479d0ae53a27c77068cb50c91/kikit/fab/jlcpcb.py#L80-L81

So, they will always appear in the POS output?

Also noticed a typo here:
https://github.com/yaqwsx/KiKit/blob/55484ff6dcc6114479d0ae53a27c77068cb50c91/kikit/fab/jlcpcb.py#L13
and here:
https://github.com/yaqwsx/KiKit/blob/55484ff6dcc6114479d0ae53a27c77068cb50c91/kikit/fab/jlcpcb.py#L26

I think "lscs" should be "lcsc"

yaqwsx, 2021-03-01:
What is the version of KiKit you use (the output of `kikit --version`)?

sslupsky, 2021-03-01:
kikit, version 0.7

yaqwsx, 2021-03-01:
The feature you are trying to use is implemented in the upstream version and wasn't released yet. This is why it is not working.

sslupsky, 2021-03-01:
Ahhh ... ok, that makes sense.

closed 2021-03-01 by yaqwsx

sslupsky, 2021-03-01:
I think you may have prematurely closed this issue.

The BOM appears to be correct now.  However, the POS file is still broken.  It includes the ignored components.  I think I pointed out the problem in one of my comments above.  You pass the entire list of components into collectPosData instead of the filtered bom.  I think the third parameter should be "bom=bom"?

sslupsky, 2021-03-01:
@yaqwsx Sorry, not bom=bom.  But, it looks to me like the lambda function always returns TRUE so all the components pass through unfiltered to posData.

yaqwsx, 2021-04-13:
@sslupsky: The POS file currently includes all components (for the sake of simplicity and not having to pair data from schematics and board). We really on the fact that all currently supported manufacturers assemble only components mentioned in BOM. We could improve it in the future but currently, there was no use case for that.