yaqwsx/KiKit #133: WSL installation doesn't work

Renha, 2021-03-02:
I followed the installation guide closely, but I get an error when typing suggested command to check the installation:
```
$ kikit --help
Traceback (most recent call last):
  File "/home/renha/.local/bin/kikit", line 5, in <module>
    from kikit.ui import cli
  File "/home/renha/.local/lib/python3.5/site-packages/kikit/ui.py", line 15
    raise RuntimeError(f"Fillet radius ({radius} mm) has to be greater than " \
                                                                            ^
SyntaxError: invalid syntax
```

yaqwsx, 2021-03-02:
There is something strange going on. The message is caused by the fact that KiKit is executed by Python 3.5, however, KiKit requires Python 3.7 (as stated in [setup.py](https://github.com/yaqwsx/KiKit/blob/99459403b5aa0a2c2ccfb660b0b9a4ae557910fb/setup.py#L13)). So I am not sure how did you manage to install it (the setup should fail in such a case).

Could you precisely describe the steps you took to install it?

Renha, 2021-03-02:
It seemed to me so too, alas I wasn't sure that was the reason. The commands were exactly the same as in installation instructions, and the command which I suppose should have had an error gave me following output:
```
 pip3 install Pcbdraw KiKit
Defaulting to user installation because normal site-packages is not writeable
Collecting Pcbdraw
  Downloading PcbDraw-0.6.1-py3-none-any.whl (361 kB)
     |████████████████████████████████| 361 kB 2.0 MB/s
Collecting KiKit
  Downloading KiKit-0.6.1-py3-none-any.whl (7.3 MB)
     |████████████████████████████████| 7.3 MB 6.4 MB/s
Requirement already satisfied: numpy in /home/renha/.local/lib/python3.5/site-packages (from Pcbdraw) (1.17.4)
Collecting lxml
  Downloading lxml-4.6.2-cp35-cp35m-manylinux1_x86_64.whl (5.5 MB)
     |████████████████████████████████| 5.5 MB 8.5 MB/s
Collecting mistune
  Downloading mistune-0.8.4-py2.py3-none-any.whl (16 kB)
Collecting wand
  Downloading Wand-0.6.6-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 11.9 MB/s
Collecting pybars3
  Downloading pybars3-0.9.7.tar.gz (29 kB)
Requirement already satisfied: pyyaml in /usr/lib/python3/dist-packages (from Pcbdraw) (3.11)
Collecting solidpython
  Downloading solidpython-0.4.0-py3-none-any.whl (64 kB)
     |████████████████████████████████| 64 kB 1.3 MB/s
Collecting markdown2
  Downloading markdown2-2.4.0-py2.py3-none-any.whl (34 kB)
Collecting shapely
  Downloading Shapely-1.7.1-cp35-cp35m-manylinux1_x86_64.whl (1.0 MB)
     |████████████████████████████████| 1.0 MB 11.9 MB/s
Collecting versioneer
  Downloading versioneer-0.18-py2.py3-none-any.whl (34 kB)
Collecting click
  Downloading click-7.1.2-py2.py3-none-any.whl (82 kB)
     |████████████████████████████████| 82 kB 868 kB/s
Collecting PyMeta3>=0.5.1
  Downloading PyMeta3-0.5.1.tar.gz (29 kB)
Collecting pypng<0.0.20,>=0.0.19
  Downloading pypng-0.0.19.tar.gz (293 kB)
     |████████████████████████████████| 293 kB 12.3 MB/s
Collecting regex<2020.0,>=2019.4
  Downloading regex-2019.12.20.tar.gz (679 kB)
     |████████████████████████████████| 679 kB 11.9 MB/s
Requirement already satisfied: PrettyTable==0.7.2 in /usr/lib/python3/dist-packages (from solidpython->KiKit) (0.7.2)
Collecting euclid3<0.2.0,>=0.1.0
  Downloading euclid3-0.01.tar.gz (13 kB)
Building wheels for collected packages: pybars3, PyMeta3, pypng, regex, euclid3
  Building wheel for pybars3 (setup.py) ... done
  Created wheel for pybars3: filename=pybars3-0.9.7-py3-none-any.whl size=12241 sha256=d72296d039afb626b43ac2a1377deb7fafc99aa8b37f33a53c7f19c49edd67b8
  Stored in directory: /home/renha/.cache/pip/wheels/21/0d/f4/e1f3ba3295947c95b39abb6c68b35d18461c1f346e51261dc5
  Building wheel for PyMeta3 (setup.py) ... done
  Created wheel for PyMeta3: filename=PyMeta3-0.5.1-py3-none-any.whl size=17599 sha256=81374ec321ba0acc1a33e9020715298283528c9410f15f25059f44ac8edb2385
  Stored in directory: /home/renha/.cache/pip/wheels/4a/17/ec/97e2e48930f89cf4cf48e84b2dae57131a2045cbbf397cdcb2
  Building wheel for pypng (setup.py) ... done
  Created wheel for pypng: filename=pypng-0.0.19-py3-none-any.whl size=53561 sha256=cb979773f5a0fb2c97b934985e22db78998ebfb7e7b077fcbc3b7fac11cc728f
  Stored in directory: /home/renha/.cache/pip/wheels/b4/8e/f8/6bd9649cca2d95353d583f8d30fdd6c57e653e71b980a9cdf1
  Building wheel for regex (setup.py) ... done
  Created wheel for regex: filename=regex-2019.12.20-cp35-cp35m-linux_x86_64.whl size=611950 sha256=20e2e35bc768cce8d3101c159d1d2d471574dcb52583132217a6f8f76dac4798
  Stored in directory: /home/renha/.cache/pip/wheels/af/69/a1/0ebcbfbbc5f75cba313693b5c1aaed0b086e0e33cf8aa406b1
  Building wheel for euclid3 (setup.py) ... done
  Created wheel for euclid3: filename=euclid3-0.1-py3-none-any.whl size=13963 sha256=bd6275c2e02f9df154d282a51a4776a59302bbc4a6670a6f1d3aad2c27a2549f
  Stored in directory: /home/renha/.cache/pip/wheels/83/47/c7/9c5af985f3ccf79faed178f0692f856bb0683316fd4849e178
Successfully built pybars3 PyMeta3 pypng regex euclid3
Installing collected packages: lxml, mistune, wand, PyMeta3, pybars3, Pcbdraw, pypng, regex, euclid3, solidpython, markdown2, shapely, versioneer, click, KiKit
Successfully installed KiKit-0.6.1 Pcbdraw-0.6.1 PyMeta3-0.5.1 click-7.1.2 euclid3-0.1 lxml-4.6.2 markdown2-2.4.0 mistune-0.8.4 pybars3-0.9.7 pypng-0.0.19 regex-2019.12.20 shapely-1.7.1 solidpython-0.4.0 versioneer-0.18 wand-0.6.6
```
May the KiKit version be also wrong?

Renha, 2021-03-02:
Ah. The `pip3` is linked to pip3.5 for me, now running with `pip3.8` command instead. I did expect it not to be able to install with the wrong version though.

yaqwsx, 2021-03-03:
Oh, I know what's going on... Before KiKit 0.7, the requirement for a minimal Python version was missing (it was added in e2b9e2966ccf7f2a7bc0fd7eff4b75d72dbc3352). It seems that pip installs the highest version that supports your version of Python. This is why KiKit 0.6.1 is installed instead of 0.7.

I will try to yank the older releases to prevent this tricky behavior.

Renha, 2021-03-03:
Thank you! The issue could be closed then I suppose, or maybe a tiny note to be aware of pip3 version on the installation page would be the full solution

yaqwsx, 2021-03-03:
Currently, you should not be able to install KiKit via Python <3.7 with a reasonable error message.

closed 2021-03-03 by Renha