yaqwsx/KiKit #140: Panel PCB for use at JLCPCB assembly service

stuartpittaway, 2021-03-19:
Enhancement request

Currently the panel and B.O.M/POS files generated by KiKit work really well.

Unfortunately JLCPCB won't accept the BOM and POS files because there are duplicate names in the "Designator" fields.

I propose that when a panel is created, each of the footprints are given a unique name, so "C1" becomes "C1_A", "C1_B" and so on.

Therefore, the BOM and POS files become useable at JLC without manual editing of the files.

yaqwsx, 2021-03-19:
I have ordered over 60 panels from JLC PCB - all generated by KiKit with no problem (the last one was the last week). There is no problem with the same designators.

Also, KiKit already supports this feature `--renameref`. Just specify the pattern.

stuartpittaway, 2021-03-19:
Thank you for the feedback, when I try to upload the BOM and CPL files today, I get "system error" from the JLC website.

I didn't see "renameref" in the documentation

stuartpittaway, 2021-03-19:
I can see #24 raises the same issue.

yaqwsx, 2021-03-19:
I would say system error isn't caused by the same references. If so, I would consider it a bug and you should report it to JLC PCB support.

stuartpittaway, 2021-03-19:
Thanks, got renameref working, thats creating a pos.csv file with "_1" and "_2" postfixes in it, however the BOM doesn't have these in, thats still the same as before.

Renha, 2021-04-01:
I've talked with their support a couple days ago, they said that same refs shouldn't be a problem. Also what is the current status of the situation for you @stuartpittaway? Did you manage to solve the issue and to place the order successfully?

yaqwsx, 2021-04-01:
The same refs aren't a problem - I ordered a panelized board just yesterday with no problems.

stuartpittaway, 2021-04-01:
Hi, thanks for investigating.  I've not progressed to an order at the moment, so if you are happy, lets close this ticket.

closed 2021-04-04 by yaqwsx