yaqwsx/KiKit #141: FramecutH and framecutV missing with mousebites and rails

bradanlane, 2021-03-20:
**Describe the bug**:
Attempting to panelize a 44mm round PCB using mousebites top/bottom and left/right between each PCB.

If I specify the panelsize then framecutH and framecutV will produce mousebites through the frame.

If I specify railsLr and railsTb then the framecutH and framecutV have no effect. This is true regardless of setting or omitting panelsize.

It would be easier to specify the tabsize, grid XY, and rails, then let KiKit compute the panelsize.

**Expected behavior**: A clear and concise description of what you expected to happen.

I was expecting framecutH and framecutV would cut through the rails. Presently, it seems framecuts are not compatible with rails. 

# To Reproduce

*I only tested with a round board as that is my current project.*

KiKit command:
This command does not cut frames
```
kikit panelize grid --space 1.6 --gridsize 2 2 --tabwidth 2.5 --tabheight 2.5 --radius 0.8 --mousebites 0.5 0.85 -0.8 --framecutV --framecutH --railsLr 5.0 --railsTb 2.0 --tooling 2.5 5.0 2.0 --fiducials 2.5 10.0 1.0 2.0 project.kicad_pcb pcbfab_panel/eccn2021panel.kicad_pcb
```
this command will cut frames
```
kikit panelize grid --space 1.6 --gridsize 2 2 --panelsize 102.8 94.8 --tabwidth 2.5 --tabheight 2.5 --radius 0.8 --mousebites 0.5 0.85 -0.8 --framecutV --framecutH --tooling 2.5 5.0 2.0 --fiducials 2.5 10.0 1.0 2.0 project.kicad_pcb pcbfab_panel/eccn2021panel.kicad_pcb
```

# Screenshots
[image]
**Panel with Rails**

[image]
**Panel with panelsize**

# Version
KiKit 0.7 (installed in Docker on Windows)

yaqwsx, 2021-03-21:
A combination of rails is not a frame - therefore, the option `--framecut` does not apply (as documented). This is the artifact of the originally simple CLI that grew over time. This will be addressed in the new CLI (as discussed in #111) and is being implemented in #139 - the part of the future v1.0 release. There won't be a fix/improvement for this in the old CLI (v0.x).

bradanlane, 2021-03-21:
Thank you for the background. I'll manually add v-scores to the generated panel.

closed 2021-03-21 by bradanlane