yaqwsx/KiKit #142: Footprint gets lost in panelization

a2800276, 2021-03-26:
**Describe the bug**: 

During panelization, a footprint of a ESP32-WROOM module is not transferred to the panelized  board. No other footprints are affected, including one for a fairly similar module (IC6).

**Expected behavior**: 
All footprints on the board should show up on the panelized version. :)

# To Reproduce

The entire project is available here: https://github.com/OpenAirCgn/openairnode-hw  
The relevant files are:
- `panelize.sh` with the invocation
- `openair2.kicad_pcb` original board
- `openair2-panelized.kicad_pcb` result of running the script

and are contained in the attached zip:
[examples.zip](https://github.com/yaqwsx/KiKit/files/6213368/examples.zip)

KiKit command:
```

$ cat panelize.sh 
# require kikit https://github.com/yaqwsx/KiKit

kikit panelize grid \
        -s 6 \
        -g 1 4 \
        --railsTb 10 \
        --railsLr 10 \
        --tabwidth 4 \
        --tabheight 4 \
        --mousebites 0.25 0.5 -0.25 \
        --tooling 5 5 4 \
        openair2.kicad_pcb \
        openair2-panelized.kicad_pcb

```

# Screenshots

See attached. The missing footprint has a large exposed pad marked "PAD GND" visible on the right of the screenshot.

[image]

# Version

```
Kikit installed via pip, python from Ubuntu apt repo

$ kikit --version
kikit, version 0.7

$ python --version
Python 3.8.5

$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal

Application: KiCad
Version: 5.1.5+dfsg1-2build2, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.8.0-45-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Technology: 7.3.0
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

```

yaqwsx, 2021-03-26:
Please read [FAQ](https://github.com/yaqwsx/KiKit/blob/master/doc/faq.md) and set `--tolerance 25`. I think it will help.

a2800276, 2021-03-27:
Perfect! Thank you. The USB port distracted me, but it's overhang is in the "Courtyard" layer not "Fab".

closed 2021-03-27 by a2800276