yaqwsx/KiKit #146: Tabs and mousebites not generating correctly

runger1101001, 2021-04-13:
I'm having great trouble finishing my panel with KiKit.

I'm using the latest docker image, and generating a panel by script. This is because it is 4 different small designs in one panel. I can place the boards and generate rails just fine, but the tabs aren't working as expected.

I'm generating the tabs from the Eco.1 layer, and depending on how I put the lines I get no tabs at all, or partial generation of tabs. When it does generate the tabs, the mousebites are often wrong.
It seems weirdly sensitive to how I place the lines. Depending on how I move them around, or the direction I draw them in, tabs get generated or not. When I move a line in one board, it sometimes affects the tabs of the other boards as well.

**Expected behavior**: Tabs and mousebites are generated correctly

I'd really appreciate some pointers what could be going wrong... what is the tab generation sensitive to? Is it important how long the lines in the Eco.1 layer are, or how they are oriented?

# To Reproduce

KiKit command:
```
#!python3
from kikit.panelize import Panel, fromMm, wxPointMM, Origin, fromDegrees, findBoundingBox
import shapely
from shapely.geometry import box
import pcbnew
import sys

startX = fromMm(50)
startY = fromMm(50)

frameThicknessMm = 5
frameThickness = fromMm(frameThicknessMm)
spacing = fromMm(3)
tabWidth = fromMm(4)

panel = Panel()

# size1, cuts = panel.makeTightGrid("../Encoder_SPI_MA730/Encoder_SPI_MA730.kicad_pcb", 1, 1, pcbnew.wxPoint(startX, startY), spacing, spacing, 
#               spacing, 0, 0, sourceArea=None, tolerance=fromMm(100), 
#               verTabWidth=tabWidth, horTabWidth=1, verTabCount=1, horTabCount=1,
#               rotation=0)
size1 = panel.appendBoard("../Encoder_AEAT_8800_Q24/Encoder_AEAT_8800_Q24.kicad_pcb", pcbnew.wxPoint(startX, startY), origin=Origin.TopLeft, tolerance=fromMm(50), rotationAngle=fromDegrees(90))
size2 = panel.appendBoard("../Encoder_SPI_A1333/Encoder_SPI_A1333.kicad_pcb", pcbnew.wxPoint(size1.x + size1.width + spacing, startY), origin=Origin.TopLeft, tolerance=fromMm(50), rotationAngle=fromDegrees(90))
size3 = panel.appendBoard("../Encoder_SPI_TLE5012B/Encoder_SPI_TLE5012B.kicad_pcb", pcbnew.wxPoint(size2.x + size2.width + spacing, startY), origin=Origin.TopLeft, tolerance=fromMm(50), rotationAngle=fromDegrees(90))
size4 = panel.appendBoard("../Encoder_SPI_MA730/Encoder_SPI_MA730.kicad_pcb", pcbnew.wxPoint(size3.x + size3.width + spacing, startY), origin=Origin.TopLeft, tolerance=fromMm(50), rotationAngle=fromDegrees(90))

gapFiller = box(size2.x, size1.y-spacing, size4.x + size4.width, size2.y-spacing)
panel.appendSubstrate(gapFiller)

minx, miny, maxx, maxy = panel.boardSubstrate.bounds()
topRail = box(minx, maxy + spacing, maxx, maxy + spacing + frameThickness)
bottomRail = box(minx, miny, maxx, miny - frameThickness)
panel.appendSubstrate(topRail)
panel.appendSubstrate(bottomRail)

tabs, cuts = panel.layerToTabs("Eco1.User", fromMm(3))
panel.appendSubstrate(tabs)
panel.makeMouseBites(cuts, diameter=fromMm(0.5), spacing=fromMm(0.9), offset=fromMm(-0.25))

panel.addMillFillets(fromMm(1))
#panel.copperFillNonBoardAreas()
panel.addFiducials(horizontalOffset=fromMm(10), verticalOffset=fromMm(2.5), copperDiameter=fromMm(1), openingDiameter=fromMm(1))
panel.addTooling(horizontalOffset=fromMm(5), verticalOffset=fromMm(2.5), diameter=fromMm(1.5))

panel.save("Encoder_SPI_Panel.kicad_pcb")
```

# Screenshots

Here's a picture of the panel it produces:
[image]

Here's a picture of the input panels:

[image]

[image]

[image]

[image]

# Version

Specify which version of KiKit do you use:
- kikit, version 0.7
- from KiKit docker image
- Docker, OSX Big Sur (Intel)

yaqwsx, 2021-04-13:
Could you provide me with the source board? Or at least show what is the result the script produces wrongly?

runger1101001, 2021-04-13:
Absolutely. The incorrect result is the top image, the combined panel - no tabs or mousebites as you can see.

The input PCBs I attach below:
[PCBs.zip](https://github.com/yaqwsx/KiKit/files/6302443/PCBs.zip)

yaqwsx, 2021-04-13:
The tabs in KiKit are not random, but they just used to be poorly documented. This is fortunatelly (partially) addressed in #139, so you can read the tab generation: https://github.com/yaqwsx/KiKit/blob/panelizeCli/doc/understandingTabs.md

So when you draw your line, you basically specify the violet line from the algorithm. Therefore, the orientation of the line matters. It has to start **inside the PCB** and end **inside the substrate**. Also, in KiKit tabs between two substrates are always generated as a two tabs starting in the same origin, with opposite direction.

Therefore, if I fix the orientation of the lines in one of your PCBs, I get:

[image]

Tabs from layers are poorly implemented. This is being reworked in #139, where I introduce annotations of the PCB. That allows you to specify the tab in a less confusing way via placement of a special footprint:

[image]

runger1101001, 2021-04-13:
It worked! Thank you so much for your rapid help!!

For anyone reading this, my problems were twofold:

1. The lines must start from outside the board, and end inside the board, as yaqwsx kindly pointed out, and as the documentation does actually say (but is a bit hard to find)
2. The lines must also start outside other substrate - in my script I had added the top/bottom rails before adding the tabs. 

After fixing the lines in the Eco.1 layer, and changing the script so tabs are generated before the outside rails everything is working great!

Here is the result:
[image]

closed 2021-04-13 by runger1101001

yaqwsx, 2021-04-13:
#139 is currently not ready for general usage as it is unstable and I push breaking changes to it quite often. Once it is semi-tested and stable I will merge it and encourage people to use.

WebDust21, 2022-03-02:
Adding to this with KiKit version 0.6.1...YES, the length of the user-drawn lines is critical as well.  If they extend too far outside the PCB boundary, KiKit will ring the entire board with a mousebite.

Spent several hours fighting over a multi-design panelized layout (as though I hadn't done it successfully several times)...only to find that drawing all the lines OUTSIDE->IN (as per above--this needs put in the documentation!) didn't fix the issue.  Had to shorten the lines so they weren't going past the mousebites...suddenly "bam" a perfect panelized board.

If the documentation clearly indicated the required direction of the lines, as well as what determines the maximum length of the line outside the board (i.e. the mousebite length)...that'd have saved me a lot of frustration.

That being said...KiKit is an extremely valuable tool--I am very grateful for it.

WebDust21, 2022-03-02:
Hmph.  Had to do a reinstall to get it working a few days ago ("autoremove" cleaned out a few too many packages), but it must not have updated the version...