yaqwsx/KiKit #151: Allow negative spacing

yaqwsx, 2021-04-18:
#139 adds generality to board placement, however, it does not allow you to specify negative offsets anymore (e.g., for oddly shaped boards to pack them more tightly). We can support it again - we just need to implement an alternative algorithm for computation of the partition line.

dmalnati, 2022-08-04:
This would be very much useful for a project I'm working on!

[image]

yaqwsx, 2022-11-28:
Arbitrary-shaped partition line could be generated as a Voronoi diagram for polygons, e.g, via https://github.com/longavailable/voronoi-diagram-for-polygons

greenscreenflicker, 2023-11-24:
You are right, #616 is a dublicate. Hence, I'd support that feature request.

TimVosch, 2024-10-03:
This might help in some cases:

https://github.com/yaqwsx/KiKit/discussions/745

[image]

usbalbin, 2025-11-28:
> Arbitrary-shaped partition line could be generated as a Voronoi diagram for polygons, e.g, via https://github.com/longavailable/voronoi-diagram-for-polygons

#874 is an attempt at doing that