yaqwsx/KiKit #152: Error: no such option: `--layout` for `kikit panelize`

jazzychad, 2021-04-19:
**Describe the bug**: Getting error `Error: no such option: --layout` when running `kikit panelize` from docker image

I read through the installation instructions in `installation.md` and installed the docker image on my local machine. I then cloned this repo and was trying to run an example panelization command from `examples.md` (see below).  I am receiving an error that `--layout` is not an option for this command, which is confusing.

**Expected behavior**: The example commands in `examples.md` should not produce error

# To Reproduce

Install the docker image: `docker pull yaqwsx/kikit`
Run docker with mount for local directory (from examples.md):
`docker run -it -w /kikit -v $(pwd):/kikit yaqwsx/kikit /bin/bash`

KiKit command:
```
# kikit panelize \
     --layout 'grid; rows: 2; cols: 2; space: 2mm' \
     --tabs 'fixed; width: 3mm; vcount: 2' \
     --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm' \
     --framing 'railstb; width: 5mm; space: 3mm;' \
     --post 'millradius: 1mm' \
 conn.kicad_pcb _mypanel.kicad_pcb

Usage: kikit panelize [OPTIONS] COMMAND [ARGS]...
Try 'kikit panelize --help' for help.

Error: no such option: --layout

# kikit panelize --help
Usage: kikit panelize [OPTIONS] COMMAND [ARGS]...

  Create a simple predefined panel patterns

Options:
  --help  Show this message and exit.

Commands:
  extractboard  Extract a single board out of a file The extracted board is...
  grid          Create a regular panel placed in a frame.
  tightgrid     Create a regular panel placed in a frame by milling a slot...
```

# Version

Specify which version of KiKit do you use:
`kikit, version 0.7`

source: `docker pull yaqwsx/kikit`

operating system: macOS 10.14.6

jazzychad, 2021-04-19:
Ohh, hmmm, looks like I have a case of very bad timing. The docs were updated only a few hours ago (looks like for tag v0.99-pre1.0) and the docker image is still v0.7

Using the examples commands in https://github.com/yaqwsx/KiKit/blob/v0.7/doc/examples.md work just fine.

So I guess the actual question is: will the docker image be updated to reflect the new tag? (I am not too familiar with docker or the images, I'm not sure how that update process works...)

yaqwsx, 2021-04-20:
Hi, this is really misfortunate timing. Having a docker with an upstream version is on the roadmap (#149). I first have to review and merge #144, then we can start building these images in CI automatically.

If you are comfortable with Docker, you can merge #144 locally and build an upstream image on your own.

jazzychad, 2021-04-20:
Ok, thank you for the update! I will see if I can incorporate the change from #144 and try it locally. I will leave it up to you if you want to close this issue or leave it open until #144 merges in case others hit this issue in the meantime.

jazzychad, 2021-04-20:
update: got the changes incorporated and built the docker image locally, and now it is running the latest code 🎉

yaqwsx, 2021-04-26:
FYI I got the nightly builds working. So you can get the latest nightly from Dockerhub via:
```
docker pull yaqwsx/kikit:nightly
```

jazzychad, 2021-04-26:
Excellent!! Thank you!! 🎉

closed 2021-07-20 by yaqwsx