yaqwsx/KiKit #153: empty space between frame and pcb

zod6, 2021-04-20:
Hi,
I couldn't find a way to have empty space (without tabs) between one side of the frame and pcb in latest git.
If tabs are added, frame spacing works. If not, frame and pcb are one piece and spacing argument is ignored.

--tabs 'fixed; hcount: 1; vcount: 0' 
[image]
--tabs 'fixed; hcount: 1; vcount: 1' 
[image]

also tried with --tabs 'fixed; hcount: 1; vcount: 1; vwidth: 0', but this gave 'divide by zero'.

full command line:
`kikit panelize --layout 'grid; rows: 3; cols: 1; space: 2mm' --tabs 'fixed; hcount: 1; vcount: 0; hwidth: 6mm' --cuts 'mousebites; drill: 0.5mm; spacing: 1mm' --framing 'frame; width: 5mm; space:2mm; cuts: false' test.kicad_pcb panel.kicad_pcb`

kikit, version 0.99-pre1.0+1.gc488420

yaqwsx, 2021-04-20:
This is indeed a bug. Should be fixed in 6384089

zod6, 2021-04-20:
Works like a charm!

closed 2021-04-20 by zod6