yaqwsx/KiKit #154: feature request: bom grouping by comment and footprint

zod6, 2021-04-20:
Hi!
it would be nice to have grouping like in 'kicost'.
Something like: if part code is missing, find component with same comment (value), footprint and non-empty partcode.

yaqwsx, 2021-04-20:
Could you be more specific? I am not sure if I understand this proposal. Could you give us an example?

zod6, 2021-04-20:
already wrote quick hack.
sorry, first time I touch python so I'm not sure if all tabs and spaces are in correct place, but it works :)

```
--- jlcpcb.py    2021-04-20 20:47:26.640284780 +0300
+++ jlcpcb.py.bak    2021-04-20 08:31:29.765568971 +0300
@@ -27,14 +27,6 @@
             orderCode = getField(c, fieldName)
             if orderCode is not None:
                 break
-
-        if orderCode is None:
-            for type, references in bom.items():
-                val, foot, lcsc = type
-                if val==getField(c, "Value") and foot==getField(c, "Footprint") and lcsc:
-                    orderCode=lcsc
-                    break
-
         cType = (
             getField(c, "Value"),
             getField(c, "Footprint"),

```

todo: traverse bom after all components are in array.

zod6, 2021-04-20:
For example I have eight identical 0.1uF 0402 capacitors.
It's quite annoying to add part number field manually to every such capacitor so it's enough to add 'lcsc' field to only one of them.

Currently:
```
0.1uF,"C3",Capacitor_SMD:C_0402_1005Metric,C1525
0.1uF,"C13,C14,C12,C15,C21,C20,C18",Capacitor_SMD:C_0402_1005Metric
```

desired otput:
`0.1uF,"C3,C13,C14,C12,C15,C21,C20,C18",Capacitor_SMD:C_0402_1005Metric,C1525`

If I must change value, for example C15, to 10uF and there are already other 10uF capacitors then I can just change it's value without worrying about part number field. Quite handy on larger boards.

yaqwsx, 2021-04-20:
You don't have to add the field individually, just open the component table in EEschema, group by value and batch enter the LCSC number info:
[image]

I am a fan of the users being explicit about their design intents so I am not a big fan of "KiKit guessing users order codes" to be honest. And batch adding them in Eeshema is really easy.

zod6, 2021-04-20:
Thank you! Didn't know that!
Probably can be closed as 'wontfix'

closed 2021-04-20 by zod6