yaqwsx/KiKit #155: Semi Circle differently shaped Edge Cut Panelization

iharshadec, 2021-04-21:
I am using KiKit Version : 0.7 on Ubuntu-Linux
Here is my PCB File with Edge Cut. 
[SVB1.zip](https://github.com/yaqwsx/KiKit/files/6349551/SVB1.zip)

Command I used to panelize this board is.

> kikit panelize grid --space 2 --gridsize 3 3 SVB1.kicad_pcb panel.kicad_pcb

Here is the error I got
```
An error occurred: SVB1.kicad_pcb: Discontinuous outline at [64.483661, 54.330586]
No output files produced
```

iharshadec, 2021-04-21:
I have checked on this point and Edge Cut seems joined nicely on this point. As when I do Spectra DSN it does not give me any error. I used that to check if my edge cut any connection error.

yaqwsx, 2021-04-21:
Your circles aren't aligned; one of them starts at 48.5221 mm and the second one on 48.5229 mm in X direction. It is a shame that KiCAD states that it uses all dimensions in nanometers in integers to be precise, but then it does not give users an option to align arcs precisely...

I guess we should make the tolerance for point snapping configurable as in many other cases, too loose tolerance causes troubles.

iharshadec, 2021-04-21:
> Your circles aren't aligned; one of them starts at 48.5221 mm and the second one on 48.5229 mm in X direction. It is a shame that KiCAD states that it uses all dimensions in nanometers in integers [...]

I see ! Let me try to adjust those values and see if that is fixing this problem or not.

iharshadec, 2021-04-21:
Damn ! It's so difficult to get those get aligned ! Not getting this fixed.

yaqwsx, 2021-04-21:
If you change -2 to -4 here: https://github.com/yaqwsx/KiKit/blob/363dc1d597a4037264d08333e24e0fb0b1815891/kikit/substrate.py#L29

It should work. I am considering making this parameter configurable as... KiCAD is not as precise as it claims to be.

iharshadec, 2021-04-21:
Didn't worked out
As I am using 0.7 version I changed this line:

[image]

Got same error Back
[image]

yaqwsx, 2021-04-26:
The imprecisely placed arcs also confused the logic that decides if the arc direction should be flipped or not (KiKit uses the convention, that the right-hand side of the outline points towards the substrate). This was addressed in 7b0eb30 - you can easily backport these changes to v0.7 if you want.

iharshadec, 2021-04-27:
> The imprecisely placed arcs also confused the logic that decides if the arc direction should be flipped or not (KiKit uses the convention, that the right-hand side of the outline points towards the [...]

I will get things updated to the latest version itself and let me test this out.

iharshadec, 2021-04-27:
@yaqwsx 
I installed latest github master copy now.

> kikit, version 0.99-pre1.0+9.gbb016f8

here is my json file
```
{
    "layout": {
        "style": "grid",
        "rows": "2",
        "cols": "2",
        "space": "3mm"
    },
	"framing": {
        "style": "railstb",
        "width": "5mm",
        "space": "3mm"
    },
    "cuts": {
        "style": "mousebites",
        "drill": "0.5mm",
        "spacing": "1mm",
        "offset": "0.2mm"
    },
    "post": {
        "millradius": "1mm"
    }	
}
```
Command

> kikit panelize -p SVB1.json SVB1.kicad_pcb panel.kicad_pcb

looks like using this way it only considers first grid configurations. there is no effect of framing or cuts or others afterwards. 
I think I am missing something in this configuration?

[image]
[image]

yaqwsx, 2021-04-27:
Instead `type` you specify `style`. I am wondering about the confusion of these two words - is it somewhere written? `style` was once on during the discussion, be I decided to use `type` as it fits most of the cases. It is possible I left `style` somewhere in the comments or documentation.

iharshadec, 2021-04-27:
> Instead `type` you specify `style`. I am wondering about the confusion of these two words - is it somewhere written? `style` was once on during the discussion, be I decided to use `type` as it fits most of the cases. It is possible I left `style` somewhere in the comments or documentation.

Yes, I was following examples.md and it has mentioned style and interesting style also works for first json type. :-) I got it corrected now and it's working.
@yaqwsx - Now KiKit is so feature rich that it needs surely a video tutorial. ;-)

yaqwsx, 2021-04-27:
Oh, the wording is fixed in 426f927.

closed 2021-04-27 by yaqwsx