yaqwsx/KiKit #160: `panelize -p <json>` doesn't work

mayl, 2021-04-23:
**Describe the bug**: new preset feature on master doesn't seem to work correctly

Running `panelize` and dumping the `json` works as expected and creates a panel.
```
❯ kikit panelize -d kikit_panel.json DRV8814_r1.kicad_pcb panel.kicad_pcb
```

Feeding the generated `json` back into `KiKit` to panelize again results in an error.
```
❯ kikit panelize -p kikit_panel.json DRV8814_r1.kicad_pcb panel.kicad_pcb
An error occurred: expected string or bytes-like object
No output files produced
```

**Expected behavior**: Running the dumped `json` through `KiKit` again should result in the same panel.

# Version
Current `master`: 93bd071

Specify which version of KiKit do you use:
Have been building `KiKit` with `Nix` [here](https://github.com/mayl/KiKit/tree/poetry_nix_master), as a first step to getting it package-able for `NixOS`.  The same build procedure, but for the tagged `v0.7` release works fine for me [here](https://github.com/mayl/KiKit/tree/poetry_nix), so I don't _think_ choking on the `json` file has to do with the build...
- the output of `kikit --version` `kikit, version 0+unknown` (If this seems strange, I should note that `Nix` disallows network access and sets date/time to a known value during builds for reproducibility reasons.  I don't know anything about `versioneer`, but I suspect there is some interaction there but have not dug in further) 
- source of installation (PyPI or upstream version) See links above to relevant branches of my fork.
- your operating system `NixOS`

yaqwsx, 2021-04-23:
This is definitely a bug. I fixed it in 0561493 and I also added pretty printing of the dumped JSON (it preserves the units entered by the user).

mayl, 2021-04-25:
Thanks, can confirm that commit fixes the issue for me so closing this.

Excited to see these `json` config files for penalization - seems like a nice compromise between writing a custom python script and using avogadro's number of command line switches.

closed 2021-04-25 by mayl