yaqwsx/KiKit #171: Nightly: 'BOARD' object has no attribute 'SetDesignSettings'

nezza, 2021-05-07:
Hello!

I am trying to run kikit with the current KiCAD nightly (2e3860de6f) on Ubuntu 20.04, and when trying to run kikit I get the following error:
```
An error occurred: 'BOARD' object has no attribute 'SetDesignSettings'
```
(I double checked that kikit is using the nightly KiCAD)

Full invocation:
```
kikit panelize \                  
    --layout 'grid; rows: 2; cols: 2; space: 2mm' \
    --tabs 'fixed; width: 5mm' \
    --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: 0.2mm' \
test.kicad_pcb panel.kicad_pcb
An error occurred: 'BOARD' object has no attribute 'SetDesignSettings'
No output files produced
```
# Version

Tried both the stable version as well as directly from HEAD (67d8f91).

yaqwsx, 2021-05-07:
This is a duplicate of #85 and #70.

Unfortunately, until the new API will be merged into nightly, there is nothing I can do. Fortunately, Seth has started working on it so hopefully, it could be ready soon (still, probably months).

closed 2021-05-07 by yaqwsx

nezza, 2021-05-07:
No worries, sorry for wasting your time!

And thank you for your great work :)