yaqwsx/KiKit #184: some THT pads missing from Cu layers

emard, 2021-06-10:
On my ulx3s project here I have made a simple Makefile
for panelization. https://github.com/emard/ulx3s

In panelized gerbers, THT Cu pad rings are missing mostly
from 2.54 mm, while other THT rings are OK.

I don't claim a bug, and welcome suggestion how
to make rings appear...

yaqwsx, 2021-06-11:
Did you read [FAQ](https://github.com/yaqwsx/KiKit/blob/v0.7/doc/faq.md#kikit-throws-away-components-from-my-panel-how-can-i-prevent-it)?

What is (probably) happening is that KiKit does not include your pin headers in the board as you have parameter tolerance too low.

emard, 2021-06-11:
I was suspecting about board origins, I haven't read FAQ,
there's no --tolarance option in v0.7 but I have tried
--sourcearea 0 0 1000 1000
to shurely include whole board but still the same - pads are missing

yaqwsx, 2021-06-11:
There is the `--tolerance` option in v0.7. I have pointed you to the documentation for v0.7. Just apply:

```
--- a/Makefile
+++ b/Makefile
@@ -20,11 +20,12 @@ RADIUS=0
 all: $(GERBERS)
 
 $(DESTINATION)/ulx3s.kicad_pcb: ulx3s.kicad_pcb $(DESTINATION)
-       $(KIKIT) panelize extractboard -s 94 51 94 112 $< $@
+       $(KIKIT) panelize extractboard -s 83 54 115 70 $< $@
 
 $(DESTINATION)/ulx3s-panel.kicad_pcb: $(DESTINATION)/ulx3s.kicad_pcb
        $(KIKIT) panelize grid     \
                --space     10     \
+               --tolerance 20 \
                --gridsize  4 2    \
                --vcuts            \
                --framecutV        \

```

Then you get what you need:

[image]
[image]
[image]

Just as described in FAQ ;)

PS: Please, in further issues, follow the issue template. It makes life easier for both of us (and it would have pointed you to the FAQ where the problem you observe is perfectly described).

emard, 2021-06-11:
> ![Screenshot from 2021-06-11
> 09-29-01](https://user-images.githubusercontent.com/1590880/121648328-af9b6000-ca97-11eb-8aa0-ebc3103b12f9.png)
> 09-28-23](https://user-images.githubusercontent.com/1590880/121648331-b0cc8d00-ca97-11eb-9456-2605e1276965.png)
> 09-27-58](https://user-images.githubusercontent.com/1590880/121648333-b0cc8d00-ca97-11eb-9800-e822a9aede24.png)

Hey I applied your suggestions and it worked!!
I was slightly confused about at which step of
the process the tolerance belongs!

Panel looks wonderful now!

yaqwsx, 2021-06-11:
The key is to read and understand how the source area is interpreted (clarified both in the documentation and FAQ).

closed 2021-06-11 by yaqwsx