yaqwsx/KiKit #187: vertical vcuts and horizontal spaces: milled regions leave spikes

emard, 2021-06-12:
This almost the result what I really need,

4 vertical v-cuts and horizontal milled spaces without
tabs. One detail is missing: at edges of board v-cuts will
leave spikes or "dents" because milling area should be
for about radius 2mm longer to the left and right side and
then board will be perfect.

I'd like suggestion how to horizontally extend milled areas
using some cmdline options?
(from project https://github.com/emard/ulx3s)

                --space     8
                --gridsize  4 2
                --vtabs     0
                --tabwidth  24
                --radius    2
                --vcuts
                --railsLr   5
                --railsTb   8
                --copperfill
                --tolerance 20
                --fiducials 5 5 1 2

[image]
[image]

yaqwsx, 2021-06-13:
The correct procedure in the KiKit workflow is to make the left and right parts of the frame thinner and add horizontal tabs.

emard, 2021-06-13:
Thanx, I added vertical tabs but now I have another problem:

horizontal mill cuts on top and bottom of the panel disappeared
and I want to keep them because now boards won't
separate from panel just by cutting at 4 vertical v-cuts

Do we have some option to make horizontal mill cuts
at top and bottom of the panel?

[image]

yaqwsx, 2021-07-20:
Hi, I am not sure if I follow. What exactly do you need to achieve? Could you hand-draw a picture?

If you would like to add a backbone, please, look at the nightly version that has a ton of new features: https://github.com/yaqwsx/KiKit/blob/master/doc/examples.md

yaqwsx, 2021-11-09:
Closing due to long inactivity. If you can fill in the missing details, feel free to reopen the issue.

closed 2021-11-09 by yaqwsx