yaqwsx/KiKit #189: Issue with Multi-part geometries

rowanfr, 2021-06-15:
**Describe the bug**:

When I use the command:
"kikit panelize -p Test.json Test2.kicad_pcb Test2Panelized.kicad_pcb"
kikit throws the error:
"An error occurred: Multi-part geometries do not provide a coordinate sequence
No output files produced"

**Expected behavior**:

I expect kikit to produce the panelized files or better inform me as to how to solve this problem as I don't know what "multi-part geometry" is referring to.

# To Reproduce

Source board and json file: 
[Test2.zip](https://github.com/yaqwsx/KiKit/files/6654128/Test2.zip)

KiKit command:
```
kikit panelize -p Test.json Test2.kicad_pcb Test2Panelized.kicad_pcb
```

Version:

- kikit, version 0.99-pre1.0+11.g2588faf
- PyPi
- Ubuntu

I have also tested this problem on:
- kikit, version 0.99-pre1.0+15.gbe2b3fc
- upstream
- Ubuntu

leo60228, 2021-06-18:
I've also run into this issue, though it was with some hacky local patches for KiCAD 6 compatibility.

leo60228, 2021-06-18:
This issue can be worked around with `--tabs fixed` instead of `--tabs full`.

yaqwsx, 2021-06-23:
The error message is unclear as it does not come from KiKit but Shapely. This message was never meant to be shown to the user; this is a bug in the tab generator (that's why other tab style helps).

rowanfr, 2021-07-06:
Looking further into this issue it is caused by the Edge.Cuts layer being used in the footprint. If I edit the footprint to delete the `Edge.Cuts` this error is removed. This error was likely not caught due to how footprints in Kicad can't use the `Edge.Cuts` layer by default and typically (in my experience) if one want to use the `Edge.Cuts` layer it has to be edited into the footprint file through the use of a text editor. 

The current working solution I have found is:
1. Edit all of the internal `Edge.Cuts` found in footprints to an **unused layer** (I used `Eco1.User` as after 2 years of using Kicad I still haven't used this and it seems to work perfectly in panelization). 
2. After this panelize the board as normal. 
3. Once it has been panelized open the `.kicad_pcb` board in Notepad ++ or your preferred text editor and replace all instances of the **unused layer** with `Edge.Cuts`. (Most text editors can replace all of them at once with Ctrl+F options)
4. Make sure after this to revert the first replacement of the **unused layer** back to the original so that this layer is still included in the layers tab off the side. This is typically within lines 14-33 in the `.kicad_pcb` file.

After all this your file should be properly panelized.

iromero91, 2021-07-13:
I have also found this issue on kicad6, workaround is as described. in my case i had holes defined in the edge.cut layer. I converted them to NPTHs.

yaqwsx, 2021-07-19:
This bug have been addressed in e1d9211. Please test it and let me know.

yaqwsx, 2021-08-23:
I did not observe further instances of the bug, closing. If you discover the problem again, feel free to reopen.

closed 2021-08-23 by yaqwsx

gwideman, 2021-08-24:
I didn't get to try the fix until today, and would like to report that it successfully deals with a board that has an internal slot defined on the Edge.Cuts layer.  Thanks for your great work on this.

However, there is one irritant -- the  --post millradius spec, which I believe is necessary to get the kikit-routed panel routed slots to have rounded ends, also operates on the existing slots. 

In my latest board, that's a slot inside the board, and another slot in the outline of the board.  I wanted the panel slots to be 2mm (so millradius 1.0mm) but that caused kikit to remove the slots that were defined by the PCB, because they are both less than 2mm.  Yet if I set millradius appropriate to the PCB-defined slots, then the ends of the panel slots aren't round.

Not fatal, but it would be nice if Kikit's process for drawing round ends on the panel slots would leave the PCB-defined slots unmolested.