yaqwsx/KiKit #191: KiKit fab does not support multi-design panels nor panels with renamed references

yaqwsx, 2021-06-23:
We should allow the user to specify multiple schematics file and match the renamed references to the original schematics.

TomKeddie, 2022-09-25:
@yaqwsx Hi Jan, I'm seeing an assertion in eeschema_v6.py when running fab on a nested design that I think is a result of this limitation.  
Digging a little deeper it seems that symbols doesn't have the parts in the subsheet but instances does.  I guess you're waiting for the eeschema api improvements to handle this but would like confirmation so I don't have to dig further.  Thank you.
```
  File "/home/tom/.local/lib/python3.10/site-packages/kikit/eeschema_v6.py", line 153, in extractComponents
    assert len(symbols) == len(instances)
AssertionError
```

yaqwsx, 2022-09-26:
Hi, this should work, and the triggered assert means that we somehow interpret the schematic file badly. This should be fixed as reading components out of sheets is a wanted functionality (and my guess is there won't be any API in the upcoming 2 years).

yaqwsx, 2022-09-27:
@TomKeddie: Could you share the schematics? I would like to reproduce the bug. If you want, you can share them confidentially via e-mail: email@honzamrazek.cz