yaqwsx/KiKit #195: Annotation Tabs Not Working

rowanfr, 2021-07-11:
**Describe the bug**: Currently when trying to panalize my board using annotation type tabs, it removes the footprint where they would be located, but doesn't generate the actual tabs.

**Expected behavior**: I expect tabs to be generated at the tip of the arrow of the footprint from kikit.pretty used to indicate the location of tabs. I expect these tabs to be generated only in between multiple boards in the grid. For the boards that have sides which are not attached to either another board or a frame, I expect tabs to not be generated on those empty sides.

# To Reproduce

Source board, output board, and json: 
[Example.zip](https://github.com/yaqwsx/KiKit/files/6796198/Example.zip)

KiKit command:
```
kikit panelize ExampleToPanelize.kicad_pcb ExamplePanelized.kicad_pcb -p Example.json
```

# Screenshots
Input Board:
[image]

Output Board:
[image]

# Version

Specify which version of KiKit do you use:
- kikit, version 0.99-pre1.0+15.gbe2b3fc
- source from PyPI
- Ubuntu

yaqwsx, 2021-07-11:
The rules for annotations are the same for any other component - they have to be included in the source area. Just increase the tolerance to e.g., 20mm:

```
{
	"source": {
		"tolerance": "20mm"
	},
	"layout": {
		"type": "grid",
		"rows": 3,
		"cols": 3,
		"hspace": "2mm",
		"vspace": "2mm"
	},
	"tabs": {
		"type": "annotation"
	}
}
```

rowanfr, 2021-07-11:
Thank you very much. Sorry about that issue. As I've used the tolerance variable before I really should have figured out that I needed to use a tolerance of 20mm.
[image]

closed 2021-07-11 by rowanfr