yaqwsx/KiKit #196: Fails:  docker pull yaqwsx/kikit:nightly

gwideman, 2021-07-12:
The bug:

docker pull kikit:nightly    fails with:

Error response from daemon: manifest for yaqwsx/kikit:nighly not found: manifest unknown: manifest unknown

I had a half hour previously successfully run 

docker pull yaqwsx/kikit:latest 

... but that only led to discovering that "latest" is a quite old version that I was trying to upgrade from.

yaqwsx, 2021-07-12:
It works flawlessly on my machine. Both in docker and podman. Please, ensure there is no problem in your docker client.

gwideman, 2021-07-12:
Thanks for responding.  The problem turns out to be with your docs at https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md  

Your doc says: 
`docker pull yaqwsx/kikit:nighly` 

... which I had copied and pasted for maximum fidelity.  But evidently it's supposed to be 
`docker pull yaqwsx/kikit:nightly` 
... which indeed works.  (And I mistakenly typed in "nightly" with a 't' when creating this issue's title and description.)

But while we are in that area of the docs, there are two other things:

`docker run -it -w /kikit -v $(pwd):/kikit yaqwsx/kikit /bin/bash`

is incorrect for Windows command line. (I'm thinking that Windows is the platform whose users will be most interested in the Docker install, since kikit can't be installed into plain Windows Kicad and Python environments.)  For Windows the command should be:

`docker run -it -w /kikit -v %cd%:/kikit yaqwsx/kikit /bin/bash`

But even that's not sufficient to invoke the nightly container. In that case we need:

`docker run -it -w /kikit -v %cd%:/kikit yaqwsx/kikit:nightly /bin/bash`

The docker run arguments have the following meanings:

- `-it`   
- `-w /kikit`     "Working directory" Sets a name as seen within docker, for the current directory
- `-v  %cd%:/kikit`   Something to do with volume that I don't understand
-   `yaqwsx/kikit:nightly`   Selects the image to run
-  `/bin/bash`  The command to run within docker.

To check which version is invoked, from the resulting bash environment we could use:

`kikit --version`

As of this writing, that command return:  
`kikit, version 0.7`   for the 'kikit:latest' container, and 
`kikit, version 0+unknown`  for the 'kikit:nightly' one.

I hope that helps others, and maybe encourages some refinement of this area of the docs.

yaqwsx, 2021-07-19:
I added the relevant bits into the documentation in 100d0dc

closed 2021-07-21 by gwideman