yaqwsx/KiKit #215: vcuts too close - appear to be on PCB cut line without allowance for vcut width

bradanlane, 2021-08-13:
**Describe the bug**: When using "full" tabs to create a panel with v-cuts between PCBs and at rails, the vcut appears to be "on" the PCB outline. When the manufacturer performs the v-score, the resulting PCB will be smaller and the clearance to the copper fill an traces will be reduced.

Within the JSON documentation, it is not clear waht the `clearance` property will do but it does not appear to change the vcuts.

**Expected behavior**: specify either the vcut width or an offset to insure the separated panel results in PCBs the correct size and with the correct clearances.

# To Reproduce

Source board: <include your `.kicad_pcb` file here>

Any simple PCB will work for this issue - even a simple _50mm square edge cut_ PCB. In my application, the PCB has a rounded corner but this is not significant to the issue.

KiKit command:
```
{
    "layout": {
        "type": "grid",
        "rows": 1,
        "cols": 1,
        "hspace": "2mm",
        "vspace": "2mm"
    },
    "source": {
        "type": "auto",
        "tolerance": "10mm"
    },
	"tabs": {
        "type": "full",
		"width": "1mm",
        "source": "none"
    },
	"cuts": {
		"type": "vcuts",
		"clearance": "10mm",
		"layer": "Cmts.User"
	},
    "framing": {
        "type": "railslr",
		"hspace": "1mm",
		"vspace": "1mm",
        "width": "10mm",
		"cuts": "v"
    },
	"tooling": {
		"type": "4hole",
		"hoffset": "3.5mm",
		"voffset": "3.5mm",
		"size": "1.8mm"
	},
	"fiducials": {
		"type": "4fid",
		"hoffset": "13.5mm",
		"voffset": "3.5mm",
		"coppersize": "1mm",
		"opening": "1.5mm"
	},
	"post": {
		"type": "auto",
		"millradius": "1mm",
		"origin": "bl",
        "copperFill": false
	}
}
```

# Screenshots

In this screen shot, the top half is the original PCB and the bottom is the top left most PCB of the panel.

[image]

# Version

kikit, version 0.99-pre1.0+33.g2e56d48

installed using: `pip3 install git+https://github.com/yaqwsx/KiKit@master`
OS: Ubuntu 21.04 (with current updates)

yaqwsx, 2021-08-13:
The clearance property adds a keep-out zone around the cut - some manufacturers do not accept V-cuts in copper and also when refilling the zones on panel, the polygons spill outside the board perimeter without the keep-out zone.

Currently, KiKit does not support cut offset for V-Cuts (it does for mousebites). However, the feature should be fairly simple to implement.

bradanlane, 2021-08-13:
OK. I manually moved the vcuts for my test. I will keep up-to-date with kikit commits.

yaqwsx, 2022-05-10:
Implemented in 68c8a02

closed 2022-05-10 by yaqwsx