yaqwsx/KiKit #233: ULX3S Cannot V-Cut a curve

emard, 2021-11-10:
I upgraded kikit development version today and
I can't build ulx3s panel with error "Cannot V-Cut a curve"

Output from older version that was giving good results
is recorded in gerbers here: 
https://github.com/emard/ulx3s-panel/tree/main/plot/gerbers

it was created with typing "make" in https://github.com/emard/ulx3s root directory)

yaqwsx, 2021-11-23:
I am sorry for missing the issue. The problem probably arises due to slight numerical instability that has been fixed in #238. In the meantime, you can work-around this issue via using: `'vcuts;cutcurves: true;'`

emard, 2021-11-24:
Yes it's numerical. I actually fixed it by increasing mill length by some
negligible amount like 0.02 mm

yaqwsx, 2021-12-19:
#238 was merged and it seems that your Makefile passes.

emard, 2021-12-19:
Thanx, I pulled latest kikit from git, run Makefile and
panel looks ok

closed 2021-12-19 by yaqwsx