yaqwsx/KiKit #246: libkicad_3dsg.so.2.0.0: cannot open shared object file

Bouni, 2021-12-16:
**Describe the bug**: 

I've installed **kicad-nightly** on Arch linux from AUR (https://aur.archlinux.org/packages/kicad-nightly/) which is then compiled from the sources which gave me version `6.0.0_rc1_477_g40d44c94cd-1`

I installed **kikit** `pip3 install git+https://github.com/yaqwsx/KiKit@master`

My kicad_pcb file was created using **kicad-nightly** on a Windows machine running version `6.0.0-rc1-452-gf844f5a80c`

If I run ` PYTHONPATH=/usr/lib/kicad-nightly/lib/python3.10/site-packages/ kikit panelize  --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts WS2812-LED-Panel.kicad_pcb WS2812-LED-Panel-2x2.kicad_pcb` I get the following error message:

```
An error occurred: libkicad_3dsg.so.2.0.0: cannot open shared object file: No such file or directory
No output files produced
```

But the `libkicad_3dsg.so.2.0.0` is there:

```zsh
cd /usr/lib/kicad-nightly/lib 
ls -la
total 592
drwxr-xr-x 4 root root   4096 Dec 15 19:08 .
drwxr-xr-x 4 root root   4096 Dec 15 19:08 ..
drwxr-xr-x 3 root root   4096 Dec 15 19:08 kicad
lrwxrwxrwx 1 root root     22 Dec 15 17:21 libkicad_3dsg.so -> libkicad_3dsg.so.2.0.0
-rwxr-xr-x 1 root root 587168 Dec 15 17:21 libkicad_3dsg.so.2.0.0
drwxr-xr-x 3 root root   4096 Dec 15 19:08 python3.10

```

**Expected behavior**: 

I expected my board to be panaelized 😏 

# To Reproduce

Source board: [WS2812-LED-Panel.zip](https://github.com/yaqwsx/KiKit/files/7724985/WS2812-LED-Panel.zip)
KiKit command: 
```
PYTHONPATH=/usr/lib/kicad-nightly/lib/python3.10/site-packages/ kikit panelize  --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts WS2812-LED-Panel.kicad_pcb WS2812-LED-Panel-2x2.kicad_pcb
```

# Version

Specify which version of KiKit do you use:
- kikit, version 0.99-pre1.0+63.g65ceb8a
- source of installation: git+https://github.com/yaqwsx/KiKit@v6support
- your operating system: Arch linux

yaqwsx, 2021-12-16:
I cannot easily reproduce this as I don't have Arch by hand. But I expect this more a packaging issue on Arch. Try setting `LD_LIBRARY_PATH` manually to the correct location (e.g., just like in https://github.com/yaqwsx/KiKit/pull/238#issuecomment-992144367)

Bouni, 2021-12-16:
🥳 YES! That did the Trick 😄 

Just for documentation purposes, here's what the command looked like:

```
LD_LIBRARY_PATH=/usr/lib/kicad-nightly/lib/ \
PYTHONPATH=/usr/lib/kicad-nightly/lib/python3.10/site-packages/ \
kikit panelize  \
--layout 'grid; rows: 2; cols: 2;' \
--tabs full \
--cuts vcuts \
WS2812-LED-Panel.kicad_pcb \
WS2812-LED-Panel-2x2.kicad_pcb
```

Thanks for the fast help!

closed 2021-12-16 by Bouni

yaqwsx, 2021-12-16:
BTW: I think you should report this to the AUR package maintainer.

Bouni, 2021-12-16:
I just had a look at the [pkgbuild](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=kicad-nightly)

`which kicad-nightly` gives me `/usr/bin/kicad-nightly` which is a shell script:

```sh
#!/bin/sh
. /usr/share/kicad-nightly/kicad-nightly.env
exec /usr/lib/kicad-nightly/bin/$prog "\$@"
EOF
```

This calls the .env file which exports `LD_LIBRARY_PATH`:

```sh
export LD_LIBRARY_PATH=/usr/lib/kicad-nightly/lib
export KICAD_PATH=/usr/share/kicad-nightly
export KICAD_CONFIG_HOME="$HOME/.config/kicadnightly"
```

This `LD_LIBRARY_PATH` is correct, so I guess something else must be the root cause?

yaqwsx, 2021-12-16:
Then it all makes sense - KiKit does not invoke `kicad-nigtly`. It just loads the corresponding `pcbnew` Python module, so with the above packaging, the LD_LIBRARY_PATH is never set. So having an env script is the way to go. We could, in theory, bake setting LD_LIBRARY_PATH into KiKit itself, but I don't want to as it is heavily distribution dependent and what you observe is only an artefact of improper packaging of the nightly version (which should not be present after stable release).