yaqwsx/KiKit #249: Kicad 6 - Generating assembly files for JLCPCB fab gets stuck in infinite loop

conradopoole, 2021-12-19:
I am using a PCB schematic saved with the latest nightly build and it seems like the extraction of components from the .pcb_sch file is getting stuck in an infinite loop.

I have tried to troubleshoot it and got all the way to sexpr.py on readSexpr(stream). It seems like its failing to parse all the expressions and at some point in one of the recursive calls it gets stuck in the while c != ")": statement....

Unfortunately I cannot figure out what the offending expression is. NOTE: My Schematic contains hierarchical sheets.

yaqwsx, 2021-12-19:
Could you share the schematics files? Without them, I cannot debug the issue.

Please, follow the issue template next time - it will guide you through all the steps I need to reproduce the issue.

conradopoole, 2021-12-19:
Just updated the description to follow the template.

Is there a way to share the schematics privately? These are for a custom job and I'm under an NDA with them.

yaqwsx, 2021-12-19:
You can e-mail them to me at email@honzamrazek.cz. I won't disclose them and I will delete them after I resolve the issue.

conradopoole, 2021-12-19:
I have found a couple syntax issue that the parser does not handle well... the \" to escape double quotes inside a double quoted string.

I removed them and it now progresses more, it still get stuck in readSexpr but parses more expressions... Looking at more "offending characters", but for now it seems that that use case of double quotes inside double quotes is not handled properly. I do not know if this is specific to Kicad 6 descriptions or it is just a coincidence in my case that they had " in my kicad6 project.

```
      (property "ki_description" "Power symbol creates a global label with name \"+3.3V\"" (id 5) (at 0 0 0)
      (property "ki_description" "Power symbol creates a global label with name \"GND\" , ground" (id 5) (at 0 0 0)
```

yaqwsx, 2021-12-19:
Could you create a minimal counterexample that features all the tricky cases?

conradopoole, 2021-12-19:
I found the rest of the issues on my files, they are due to the backslashes present in double quoted values. In a couple cases was due to the \ being used to escape the ", and in the other cases it was the \ in \n to introduce a new line in text blocks that I use to add comments in my schematics.
```
(property "ki_description" "Power symbol creates a global label with name \"+3.3V\"" (id 5) (at 0 0 0)
(property "ki_description" "Power symbol creates a global label with name \"GND\" , ground" (id 5) (at 0 0 0)

 (text "     Auto program \n\nDTR RTS  —>  EN IO0 \n1     1        1    1\n0     0        1    1 \n1     0        0    1\n0     1        1    0"
    (at 59.69 33.02 0)
    (effects (font (size 1.27 1.27)) (justify left bottom))
    (uuid bfd87374-c052-4c24-b269-ac9427811e17)
)

(text "RJ45 Plug termination \ncompatible with T568A or T568B"
    (at 33.02 21.59 0)
    (effects (font (size 1.27 1.27)) (justify left bottom))
    (uuid 98265c5a-5db5-4fbe-9ac0-2adb6b10b08e)
)

(text "Differential LED Outputs\n" (at 14.732 17.653 0)
    (effects (font (size 1.27 1.27)) (justify left bottom))
    (uuid 4a838bb9-3dfb-4692-abdb-a823328d3d6c)
)
```

After removing these \n and the \" the kikit command completes succesfully.

```
--- Parsing 0.0713186264038086 seconds ---
Got sheet!
--- Parsing 0.051392555236816406 seconds ---
Got sheet!
--- Parsing 0.007995843887329102 seconds ---
Got sheet!
--- Parsing 0.02418231964111328 seconds ---
Got sheet!
--- Parsing 0.016138076782226562 seconds ---
Got sheet!
--- Parsing 0.009213685989379883 seconds ---
Got sheet!
--- Parsing 0.03281855583190918 seconds ---
Got sheet!
--- Parsing 0.03445863723754883 seconds ---
Got sheet!
--- Parsing 0.016440629959106445 seconds ---
Got sheet!
--- Parsing 0.005858421325683594 seconds ---
Got sheet!
--- Parsing 0.028935670852661133 seconds ---
WARNING: Component Y1 is missing ordercode
```

I will create a simple project with an schematic and a pub to demonstrate this easily. 

I have also checked and it looks like all Kicad 6 Power symbols use double quotes around the value so this will definitely become an issue.

[image]

yaqwsx, 2021-12-19:
Once you create a project that triggers this behavior, I will look at it.

conradopoole, 2021-12-19:
@yaqwsx Here is a sample project that triggers the issue, with just a barrel connector, a description text with new lines and GND and +3V3 symbols

[Test_Kikit_#249.zip](https://github.com/yaqwsx/KiKit/files/7741247/Test_Kikit_.249.zip)
```
kikit fab jlcpcb --assembly --schematic Test_Kikit_#249.kicad_sch Test_Kikit_#249.kicad_pcb ./testfab
```

You'll see that it gets stuck after the Gerber generation when it starts starting the Sexpr. If you edit the PCB file with a text editor and get rid of the \\" and  \\n then it starts working.

conradopoole, 2021-12-19:
I believe this PR might fix the issue #250

When detecting the \ escape character the stream was not being read and it would not advance, getting stuck peeking the \ over and over again.

```
def readQuotedString(stream):
    stream.shift('"')
    s = []
    escaped = False
    c = stream.peek()
    while c != '"' or escaped:
        if c == "\\":
            escaped = True
            stream.read() ## Added this line to advance the stream
        else:
            escaped = False
            s.append(stream.read())
        c = stream.peek()
    stream.shift('"')
    return "".join(s)
```

closed 2021-12-21 by yaqwsx