yaqwsx/KiKit #251: DLL load failed while importing _pcbnew

AiGreek, 2021-12-19:
```
Application: Editeur de PCB KiCad (64-bit)

Version: (6.0.0-rc1-402-gcaf406943e), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19043), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 11 2021 21:00:06
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```
After trying to install Kikit with some difficulties, and following the installation tutorial [here](https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md) I find myself stuck when trying to activate the plugin(s) in kicad.
[image]
I think I installed as many dependencies as possible, but this error persist. Did I miss something?

yaqwsx, 2021-12-21:
Did you try it with the latest nightly as we discussed on Discord?

AiGreek, 2021-12-21:
Yes sorry, I installed the nightly version released yesterday but I didn't have time to test (4h of download ................). I tried just now, and all the commands work correctly as well as the plugins in Kicad.

However, I have no idea why / how it works now and not at the time of creating this issue.

[image]

yaqwsx, 2021-12-22:
Ok, so the problem was resolved, right?

AiGreek, 2021-12-22:
Yep, it is ... for now ^^ 
thanks

closed 2021-12-22 by AiGreek