yaqwsx/KiKit #252: v6 MacOS functionality and wxwidgets

a7hybnj2, 2021-12-20:
I just updated kicad and kikit. I am running
`KiCad Version: (6.0.0-rc2-14-ga17a58203b), release build` and
`kikit, version 0.99-pre1.0+76.g2d499e0` although I tired this with 0.7 also. On
`macOS 12.0.1`

The error I get is
```
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/libwx_osx_cocoau-3.1.5.dylib
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/local/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file), '/usr/lib/libwx_osx_cocoau-3.1.5.dylib' (no such file)
No output files produced
```
This was referenced in #21 but this is indeed a different issue.

I was able to get past the initial error that was looking for `libwx_osx_cocoau_gl-3.1.5.dylib` by installing `wxwidgets` using `brew install wxwidgets` but now that I think about it that shouldn't have been necessary. It shouldn't be looking for the local libs...right? Shouldn't all of that be packaged into the kicad environment?

I know yaqwsx doesn't have a mac at their disposal so I am just putting my notes here in case they or anyone else gets a similar error or figures how to progress past this point.

yaqwsx, 2021-12-21:
I am not a mac user but I was wondering if the binary `pcbnew-nightly isn't just a wrapper script that properly set all library paths - just like in #246. The paths, of course, could be different.

a7hybnj2, 2021-12-21:
So I set `LD_LIBRARY_PATH` to `/Applications/KiCad/KiCad.app/Contents/Frameworks/` which is where
```
libwx_osx_cocoau_gl-3.1.5.0.0.dylib
libwx_osx_cocoau_gl-3.1.5.dylib
libwx_osx_cocoau-3.1.5.0.0.dylib
libwx_osx_cocoau-3.1.5.dylib
```
all live. But, I still got the same error as above. Is there another path I should be setting?

yaqwsx, 2021-12-21:
Quick googling told me that mac uses `DYLD_LIBRARY_PATH` instead of `LD_LIBRARY_PATH`. Try that. But I am shooting blindly as I have no experience with mac...

a7hybnj2, 2021-12-21:
Cool, one step closer. After setting `DYLD_LIBRARY_PATH`
```
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/Versions/3.8/Python
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python' (no such file), '/usr/local/lib/Versions/3.8/Python' (no such file), '/usr/local/lib/Versions/3.8/Python' (no such file), '/usr/local/lib/Python' (no such file), '/usr/lib/Python' (no such file)
No output files produced
```
So the `Python` it is looking for is in `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/` and the PYTHONPATH is pointing there... Seems like there might be another thing to set...

a7hybnj2, 2021-12-21:
Okay I set my `DYLD_LIBRARY_PATH` to `/Applications/KiCad/KiCad.app/Contents/Frameworks` because that is where the `libwx_osx_cocoau` files live. That appears to have resolved that problem but now it cannot locate "Python" itself.
```
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/Versions/3.8/Python
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python' (no such file), '/usr/local/lib/Versions/3.8/Python' (no such file), '/usr/local/lib/Versions/3.8/Python' (no such file), '/usr/local/lib/Python' (no such file), '/usr/lib/Python' (no such file)
```

yaqwsx, 2021-12-22:
Could you:
- check content `DYLD_LIBRARY_PATH` before setting a value?
- check what does the `pcbnew-nighlty` binary does. I guess it will be a shell script

yaqwsx, 2021-12-22:
BTW: I am sorry; I probably closed this issue yesterday by mistake when posting an answer.

cdwilson, 2021-12-22:
@yaqwsx I think the issue here is that the Python bundled with KiCad is ultimately calling `ctypes.util.find_library()` which probably can't find any `*.dylib` installed by homebrew.  Homebrew patches Python to fix this (see https://github.com/Homebrew/homebrew-core/blob/15e7f288e4b97c75b32c75748df0daa3b60d1c77/Formula/python%403.9.rb#L185), but I'm guessing the KiCad devs aren't patching the Python included with KiCad in the same way.

@a7hybnj2 Can you try adding something like the following to your `~/Library/Preferences/kicad/6.0/PyShell_pcbnew_startup.py`:
```
import sys
if sys.platform == "darwin":
    import os
    os.environ["DYLD_FALLBACK_LIBRARY_PATH"] = "/opt/homebrew/lib/" <-- put your path to homebrew here
```

This should allow the KiCad python to find any dylibs installed by homebrew.

[image]

[image]

cdwilson, 2021-12-23:
> I was able to get past the initial error that was looking for libwx_osx_cocoau_gl-3.1.5.dylib by installing wxwidgets using brew install wxwidgets but now that I think about it that shouldn't have [...]

Sorry, I misread this earlier, and now that I think about this, I think @a7hybnj2 is right, it seems like these should be included with KiCad in `/Applications/KiCad/KiCad.app/Contents/Frameworks`, not something installed by homebrew.

> Okay I set my DYLD_LIBRARY_PATH to /Applications/KiCad/KiCad.app/Contents/Frameworks because that is where the libwx_osx_cocoau files live. That appears to have resolved that problem but now it cannot locate "Python" itself.

Slight edit to what I proposed earlier, this should allow KiCad Python to find `libwx_osx_cocoau_gl-3.1.5.dylib`:

```
# ~/Library/Preferences/kicad/6.0/PyShell_pcbnew_startup.py
import sys
if sys.platform == "darwin":
    import os
    os.environ["DYLD_FALLBACK_LIBRARY_PATH"] = "/Applications/KiCad/KiCad.app/Contents/Frameworks"
```

[image]

a7hybnj2, 2021-12-23:
I tested with your above fix and was still getting the same errors so just in console I
```
export PYTHONPATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ kikit
export DYLD_LIBRARY_PATH=/Applications/KiCad/KiCad.app/Contents/Frameworks
export DYLD_FALLBACK_LIBRARY_PATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8
```
all of the previous errors went away and were replaced by: `Segmentation fault: 11`

What is `@rpath`? after setting just PYTHONPATH and the FALLBACK it is trying to work but `@rpath/Versions/3.8/Python` doesn't expand correctly. `An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/Versions/3.8/Python` if @rpath=Python.framework then it would be looking for python in the right spot...

cdwilson, 2021-12-23:
I'm brand new to KiKit, and I didn't realize earlier that it wasn't just using the Python bundled with KiCad, so the `PyShell_pcbnew_startup.py` change I mentioned above won't make any difference running `kikit` externally.  I tried going through the install instructions last night, but I wasn't able to get it working on my system either.

@a7hybnj2 are you seeing any difference when setting `DYLD_LIBRARY_PATH` or `DYLD_FALLBACK_LIBRARY_PATH` on the command line?  Passing those on the command line doesn't seem to be working for me (it appears that SIP might be sanitizing `DYLD_*` environment variables, see https://briandfoy.github.io/macos-s-system-integrity-protection-sanitizes-your-environment/):

```
$ PYTHONPATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ \
DYLD_FALLBACK_LIBRARY_PATH=/Applications/KiCad/KiCad.app/Contents/Frameworks \
kikit panelize --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts doc/resources/conn.kicad_pcb panel.kicad_pcb
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/libwx_osx_cocoau_gl-3.1.5.dylib
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/usr/local/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file), '/usr/lib/libwx_osx_cocoau_gl-3.1.5.dylib' (no such file)
No output files produced
```

> What is @rpath?

It appears to be a macro for the dynamic loader that lets you specify dependency paths that are resolved at runtime.  (see https://developer.apple.com/library/archive/documentation/DeveloperTools/Conceptual/DynamicLibraries/100-Articles/RunpathDependentLibraries.html#//apple_ref/doc/uid/TP40008306-SW1)

> When the dynamic loader (dyld) loads the executable, it looks for run-path dependent libraries in the run-path search paths in the order in which they were specified at link time

You can inspect these rpaths using `otool`:
```
$ $ otool -L /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so:
        /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
        /System/Library/Frameworks/Carbon.framework/Versions/A/Carbon (compatibility version 2.0.0, current version 164.0.0)
        /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa (compatibility version 1.0.0, current version 23.0.0)
        /System/Library/Frameworks/AudioToolbox.framework/Versions/A/AudioToolbox (compatibility version 1.0.0, current version 1000.0.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.60.1)
        /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
        @rpath/libwx_osx_cocoau_gl-3.1.5.dylib (compatibility version 1.0.0, current version 1.0.0)
        @rpath/libwx_osx_cocoau-3.1.5.dylib (compatibility version 1.0.0, current version 1.0.0)
        @rpath/Versions/3.8/Python (compatibility version 3.8.0, current version 3.8.0)
        @rpath/libkicad_3dsg.2.0.0.dylib (compatibility version 2.0.0, current version 0.0.0)
        /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit (compatibility version 45.0.0, current version 2022.20.117)
        /System/Library/Frameworks/CoreData.framework/Versions/A/CoreData (compatibility version 1.0.0, current version 1044.3.0)
        /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation (compatibility version 300.0.0, current version 1770.255.0)
        /usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
        @rpath/libGLEW.2.2.dylib (compatibility version 2.2.0, current version 2.2.0)
        @rpath/libcairo.2.dylib (compatibility version 11603.0.0, current version 11603.0.0)
        @rpath/libpixman-1.0.dylib (compatibility version 41.0.0, current version 41.0.0)
        /usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 904.4.0)
        /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation (compatibility version 150.0.0, current version 1770.255.0)
        /usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
```

You can actually experiment with changing the rpaths using `install_name_tool`:
```
install_name_tool -change @rpath/Versions/3.8/Python /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Python /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
install_name_tool -change @rpath/libwx_osx_cocoau_gl-3.1.5.dylib /Applications/KiCad/KiCad.app/Contents/Frameworks/libwx_osx_cocoau_gl-3.1.5.dylib /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
install_name_tool -change @rpath/libwx_osx_cocoau-3.1.5.dylib /Applications/KiCad/KiCad.app/Contents/Frameworks/libwx_osx_cocoau-3.1.5.dylib /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
[... cut ...]
        /usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
```

However, even after replacing the rpaths with absolute paths, I'm still getting the same `Segmentation fault: 11` as @a7hybnj2:

```
$ PYTHONPATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ \
kikit panelize --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts doc/resources/conn.kicad_pcb panel.kicad_pcb
Segmentation fault: 11
```

At this point, I'm out of ideas... 🤷

a7hybnj2, 2021-12-24:
Now that KiCad 6 is officially out I suspect it is only a matter of time before someone figures it out. I too have done all I know how to do.

yaqwsx, 2021-12-24:
I am still unsure whether it is safe to release v1, if I cannot confirm it works on Mac. Or if it is better to release v1, find the Mac users and then release v1.0.1 that is patched for Mac.

cdwilson, 2021-12-26:
Given that Mac users can run KiKit in Docker, I'd say just release v1 and make a note that macOS isn't supported yet (who knows how long until somebody figures out how to get this working on Mac...)

tywtyw2002, 2021-12-29:
It seems like the MacOS Version KiCAD compile a standalone python version. The python library did not share w/ system or brew version python.

The easy way is to use kicad python to install kikit. And use that kikit to do panelize.

**Install**
`/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/pip3 install kikit`

**Panelize**
` /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 1;' --tabs full  --cuts vcuts ./example.kicad_pcb panel.kicad_pcb
`

a7hybnj2, 2021-12-29:
Well, I just tried the above and it seems promising although I still ran into some errors. When I installed kikit I get a
`ERROR: wxpython 4.1.1 requires pillow, which is not installed.`
so I just did a 
`/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/pip3 install pillow`
but then when I ran the panelize command I got
```
An error occurred: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3.8"
  * The NumPy version is: "1.21.5"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so, 0x0002): tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so' (code signature in <D6DEC3D7-AF4F-3AA2-B2DD-D6A87AC638BA> '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so' not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)), '/usr/lib/_multiarray_umath.cpython-38-darwin.so' (no such file)
```
seems like this might be a pythonpath dyld thing again. I will try to locate those packages and point to them correctly and update here.

tywtyw2002, 2021-12-29:
You got this error `ERROR: wxpython 4.1.1 requires pillow, which is not installed.` That fine. Skip it.

a7hybnj2, 2021-12-29:
`/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/pip3 -V
pip 19.2.3 from /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/pip (python 3.8)`
```
echo $PYTHONPATH

```
(The above is a blank line)
I removed my kicad install including everything in ~/Library/Preferences and ~/Library/Caches and tried again without installing pillow this time:
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/pip3 install kikit
Collecting kikit
  Using cached https://files.pythonhosted.org/packages/e2/6a/b67074cc94d4373c7e82b661c59a227fa51fb5251e96a721d375cb91e49a/KiKit-1.0.0-py3-none-any.whl
Collecting shapely (from kikit)
  Using cached https://files.pythonhosted.org/packages/80/4f/d96da54dadc06b3202df977f3721cc72f9f322f0ec5f15dd0baabd5ab365/Shapely-1.8.0-cp38-cp38-macosx_10_9_x86_64.whl
Collecting numpy (from kikit)
  Using cached https://files.pythonhosted.org/packages/e8/b2/be38f1036c7e7f6af79201f09de86615599beeb232f37dd288d9631d9b29/numpy-1.21.5-cp38-cp38-macosx_10_9_x86_64.whl
Collecting click (from kikit)
  Using cached https://files.pythonhosted.org/packages/48/58/c8aa6a8e62cc75f39fee1092c45d6b6ba684122697d7ce7d53f64f98a129/click-8.0.3-py3-none-any.whl
Collecting solidpython (from kikit)
  Using cached https://files.pythonhosted.org/packages/b6/1e/3a065797bd8fe8e2e0ced73e6baf07fdb921fecc04fa18b2b8ffb34f3c87/solidpython-1.1.1-py3-none-any.whl
Collecting markdown2 (from kikit)
  Using cached https://files.pythonhosted.org/packages/ab/36/101b2a0fb8880ae7ea2bbcee5a2f91e74bb58e5f23d0d692020fb07623c6/markdown2-2.4.2-py2.py3-none-any.whl
Collecting commentjson (from kikit)
Collecting pybars3 (from kikit)
Collecting pcbnewTransition==0.2.0 (from kikit)
  Using cached https://files.pythonhosted.org/packages/98/cc/2210a2dca06d89c4f87e1c2b6e10d20e8a4d32b9fbe1590e9d188b3123bd/pcbnewTransition-0.2.0-py3-none-any.whl
Collecting euclid3<0.2.0,>=0.1.0 (from solidpython->kikit)
Collecting pypng<0.0.20,>=0.0.19 (from solidpython->kikit)
Collecting regex<2020.0,>=2019.4 (from solidpython->kikit)
Collecting PrettyTable==0.7.2 (from solidpython->kikit)
Collecting lark-parser<0.8.0,>=0.7.1 (from commentjson->kikit)
Collecting PyMeta3>=0.5.1 (from pybars3->kikit)
ERROR: wxpython 4.1.1 requires pillow, which is not installed.
Installing collected packages: shapely, numpy, click, euclid3, pypng, regex, PrettyTable, solidpython, markdown2, lark-parser, commentjson, PyMeta3, pybars3, pcbnewTransition, kikit
Successfully installed PrettyTable-0.7.2 PyMeta3-0.5.1 click-8.0.3 commentjson-0.9.0 euclid3-0.1 kikit-1.0.0 lark-parser-0.7.8 markdown2-2.4.2 numpy-1.21.5 pcbnewTransition-0.2.0 pybars3-0.9.7 pypng-0.0.19 regex-2019.12.20 shapely-1.8.0 solidpython-1.1.1
WARNING: You are using pip version 19.2.3, however version 21.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
looks just like yours
then
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 1;' --tabs full --cuts vcuts conn.kicad_pcb panel.kicad_pcb
An error occurred: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3.8"
  * The NumPy version is: "1.21.5"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so, 0x0002): tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so' (code signature in <D6DEC3D7-AF4F-3AA2-B2DD-D6A87AC638BA> '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so' not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)), '/usr/lib/_multiarray_umath.cpython-38-darwin.so' (no such file)

No output files produced
```
I am using the example board provided by kikit in the doc/resources folder. Could it be the example file? Do you have a file you want to share to rule that out?

yaqwsx, 2021-12-29:
> I am using the example board provided by kikit in the doc/resources folder. Could it be the example file? Do you have a file you want to share to rule that out?

Once you panelize, `numpy` is imported. So it won't be just this case, but every case. Actually, help imports only `click` and nothing else (not even the `pcbnew` module - as it takes quite a long time to load).

tywtyw2002, 2021-12-30:
This is MacOS Standard security issue..... signature error... 

I do not know how to resolve it. (Long long time ago, I disabled lots of macos security options.)

Hits: search google `mapped file has no Team ID and is not a platform binary`

```
Original error was: 

dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so, 0x0002): 

tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so' 

(code signature in <D6DEC3D7-AF4F-3AA2-B2DD-D6A87AC638BA> 

'/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/numpy/core/_multiarray_umath.cpython-38-darwin.so'

 not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)),

 '/usr/lib/_multiarray_umath.cpython-38-darwin.so' (no such file)

```

conradopoole, 2021-12-30:
I managed to get the SIP (System Integrity Protection) Mac issue solved, but I had to sign again the Python.app embedded with Kicad.

If you don't have a Apple Developer code signing certificate you can simply create a Self Signed using Keychain like this...

[image]

Then run 

```
codesign -fs "my-certificate" /Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app
```

I was then able to run using the embedded python the example panelization command...

```
/Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 1;' --tabs full --cuts vcuts <PATH_TO_PCB>/conn.kicad_pcb <OUTPUT_PATH>/panel.kicad_pcb
```

Technically I believe this should be solved by the Kicad Mac builder, they should probably add the "com.apple.security.cs.disable-library-validation=TRUE" entitlement the the Python.app. The problem is that the Kicad.app is code signed but the minute we run the embedded pip3, files are added that were not part of the original bundle so they cannot be loaded. The issue is solved by resigning the Python.app but ideally we'd like that validation to be disabled.

Summarizing, install KiKit, code sign and then run kikit.

```
/Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/pip3 install kikit
codesign -fs "my-certificate" 

/Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app

/Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 1;' --tabs full --cuts vcuts <PATH_TO_PCB>/conn.kicad_pcb <OUTPUT_PATH>/panel.kicad_pcb
```

conradopoole, 2021-12-30:
I forgot to mention that even though the panelize command runs and generates the panel, it also outputs some assertion issues:

```
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
```

tywtyw2002, 2021-12-30:
I got same outputs too, but it seems not affect the panelize. It only the message for wxApp.

> I forgot to mention that even though the panelize command runs and generates the panel, it also outputs some assertion issues:
>
> ```
> ./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): [...]

a7hybnj2, 2021-12-30:
@conradopoole sweet! I will try this out this afternoon.

Would this be mitigated if this package were installed through the new kicad plugin/package/theme navigator? I know that isn't really how this tool works but would be cool and much easier for non-cl people.

cdwilson, 2021-12-30:
> When I installed kikit I get a `ERROR: wxpython 4.1.1 requires pillow, which is not installed.`

@a7hybnj2 I think you can prevent the error from showing up by passing the `--no-deps` flag to pip when installing `numpy` and `pillow` (this installs for me without any errors):

```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip install --upgrade pip
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip --no-deps install --upgrade numpy pillow
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip install --upgrade kikit
```

@conradopoole the code signing works for me too.  I'm also getting the assertion errors, but it looks like the panelizer is working correctly.

```
$ /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 2;' --tabs full --cuts vcuts doc/resources/conn.kicad_pcb panel.kicad_pcb
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
```

@yaqwsx FYI, when I open pcbnew after installing the KiKit front end from PCM, I get the following error.  What is it looking for to determine if there is a backend installed?

[image]

tywtyw2002, 2021-12-31:
@cdwilson On my side, the kicad kikit plugin work perfect.

@yaqwsx For macos, when you enable the plugin, the script should in `/Users/xxx/Documents/kicad/6.0/scripting/plugins/` instead of `/Users/xxxx/.local/share/kicad/6.0/scripting/`.

Following steps

- enable plugin by kikit-plugin
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit-plugin enable panelize hideReferences
File '/Users/xxxx/.local/share/kicad/6.0/scripting/kikit_plugin.py' was created
```
- Mv kikit_plugin to kicad plugin folder.
```
mv /Users/xxxx/.local/share/kicad/6.0/scripting/kikit_plugin.py /Users/xxx/Documents/kicad/6.0/scripting/plugins/kikit_plugin.py
```

PS: Fingers crossed and good luck

conradopoole, 2021-12-31:
@tywtyw2002 Did you do any additional code signs? For me the Panelize plugin does not work either from the GUI (it does not load with same message as @cdwilson ) unless a fully code sign the full Kicad.app (with --deep), then it loads

```
codesign --deep -fs "my-certificate" /Applications/Kicad/KiCad.app
```

tywtyw2002, 2021-12-31:
NO, I disabled SIP and other security features since day 1.

conradopoole, 2021-12-31:
> NO, I disabled SIP and other security features since day 1.

Ah, I see, I never disabled it, I like those security features...

tywtyw2002, 2021-12-31:
> > NO, I disabled SIP and other security features since day 1.
> 
> Ah, I see, I never disabled it, I like those security features...

Maybe, popup an issue to kicad issue tracker to let them enabled the 3-party binary execute permission.

tywtyw2002, 2021-12-31:
Try resign the 

`/Applications/KiCad/KiCad.app/Contents/Applications/pcbnew.app`

yaqwsx, 2022-01-02:
@cdwilson: The message you see means that KiKit is not installed for the Python interpreter Pcbnew uses. It is shown when `from kikit.plugin import importAllPlugins` fails.

@tywtyw2002  Those paths are on what platform? macOS? I am not sure if I want to fix this as we have KiCAD PCM which is much user-friendlier. The `kikit-plugin` was created as there was no easy way of enabling the plugins. However, maintaining it means walking through KiCAD sources and replicating the path-building functions as this part of KiCAD is not exported in the Python API.

tywtyw2002, 2022-01-04:
> /Users/xxx/Documents/kicad/6.0/scripting/plugins/

Yep, on Macos, the default script path is `/Users/xxx/Documents/kicad/6.0/scripting/plugins/`

yaqwsx, 2022-01-05:
BTW: I opened a feature request in KiCAD mac package that could solve this: https://gitlab.com/kicad/packaging/kicad-mac-builder/-/issues/384

Gasman2014, 2022-01-07:
FWIW I can confirm that resigning the Kicad.app __does__ work - it may only be necessary to sign the pcbnew.app but I din't try that. I have not needed to disable SIP.

Robomaticii, 2022-01-27:
Is there a definitive method for getting Kikit to work on Macs? I get the same notification about not installing the backend. Well... I have installed the backend and I get a help listing etc. I also installed the frontend from the Plugins area. However they just don't connect. It is a issue with t an incorrect path?

Gasman2014, 2022-01-27:
I got it to work by signing the Kicad app - as per @tywtyw2002 instructions above. i.e generate your own certificate and then do

```codesign --deep -fs "my-certificate" /Applications/Kicad/KiCad.app```

and 

```mv /Users/xxxx/.local/share/kicad/6.0/scripting/kikit_plugin.py /Users/xxx/Documents/kicad/6.0/scripting/plugins/kikit_plugin.py```

Robomaticii, 2022-01-28:
John

Thanks for following up with me.

However, I can't find "kikit_plugin.py."

I installed the backend using: pip3 install kikit

When I type "kikit --help " it works. When I type "kikit --version" it
works.

I've installed Kikit from the PCM and it says it is installed.

When I look for kikit_plugin.py in
/Users/xxxx/.local/share/kicad/6.0/scripting/kikit_plugin.py
(or anywhere else) I just can't find it.

If I can't find it I can't move it to
/Users/xxx/Documents/kicad/6.0/scripting/plugins.

Any help would be appreciated.

Thanks

Robert AKA Robomaticii

Gasman2014, 2022-01-28:
I think this is what I needed to do too get it working.

 /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip install --upgrade pip numpy pillow kikit
 /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit-plugin enable panelize hideReferences
 codesign -fs “Your Cert" /Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app
 mv $HOME/.local/share/kicad/6.0/scripting/kikit_plugin.py $HOME/Documents/KiCad/6.0/scripting/plugins/

YMMV - This was on Monterey. You need to do the plugin-enable to generate the plugins before they can be moved.

Robomaticii, 2022-01-29:
John

Thanks for the updated info.

However, I haven't been able to get past the first line of instructions. It
may be that all my attempts have messed things up to the point it is not
possible.

I am going to deinstall KiCad and do a fresh install. This may allow me to
get it done.

Regards

Robert

Gasman2014, 2022-01-30:
You have to type the whole line as one function

/Applications/KiCad/…../python3 -m pip install —upgrade …

ALL as one line - the forum software is splitting it into two lines. 

John Pateman

Robomaticii, 2022-01-30:
John

My mistake on the first line was a little more foolish. I copied and pasted
into Terminal without the first "/" before Application. Too easy to miss.

Now I am having problems with the following  instruction lines.

I still can't find "kikit_plugin.py".

I am on Catalins. This may be an OS version thing. I will update to
Monterey and see whether that changes anything.

Thanks again for following up with me.

Robert

Gasman2014, 2022-01-31:
I have successfully installed on Monterey but would be very surprised if Catalina was any more problematic. In fact, system security has got tighter in more recent os iterations so Catalina might well be easier.

What happens if you just evoke the python interpreter on its own - just check that you get a console running by typing

```/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3```

into a terminal window. You should get a python console with a prompt. (Type exit() to quit this).

This relies on you having installed KiCad in tot he normal Applications directory. If that doesn’t work, I would try installing the application again and then running the full incantations once more.

> On 30 Jan 2022, at 17:42, Robomaticii ***@***.***> wrote:
> 
> /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3

themaninthesuitcase, 2022-01-31:
> I think this is what I needed to do too get it working. /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip install --upgrade pip numpy pillow kikit [...]

I just tried this on Big Sur and needed to sign Python.app before I could run kikit-plugin enable.  Maybe this should be the 1st step.

It all ran successfully from terminal but still seeing the "No backend found" error when I open pcbnew.  

If I run say `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit fab [params]` it works.  It moans `assert ""traits"" failed in Get(): create wxApp before calling this` a few times but there's a zip where I expect it.

conradopoole, 2022-01-31:
> I just tried this on Big Sur and needed to sign Python.app before I could run kikit-plugin enable. Maybe this should be the 1st step.
>
> It all ran successfully from terminal but still seeing the "No [...]

Did you try on pcbnew when launched from a Kicad window? Or standalone? In my experience it only works when I launch Pcbnew.app standalone, not from Kicad. When I do it from Kicad I always get that issue with the "No backend found"

Robomaticii, 2022-02-02:
John

So I believe I have got KiKit to install properly so that it works within
KiCad. I haven't used it yet but it is available in Pcbnew and opens a menu
when you click on the icon.

I started with a fresh install of KiCad.

I also generated my certificate as suggested in Issue #252.

Using the certificate I signed Python.app as follows:

codesign -fs “my-certificate"
/Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app

Next step is as you suggested:

/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3
-m pip install --upgrade pip numpy pillow kikit

and:

/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit-plugin
enable panelize hideReferences

I made a plugins" directory in /Users/"User
Name"/Documents/kicad/6.0/scripting/

Then I opened KiCad and installed KiKit from the PCM.

Close KiCad.

Check to see if kikit_plugin.py is in
/Users/chimney/Documents/kicad/6.0/scripting/plugins/   - it was there for
me.

If not, it would have to be moved there from wherever.

Last step is to sign the KiCad.app.

codesign -fs my-certificate /Applications/Kicad/KiCad.app

Voilà. C'est ça.

I will post it on Issue #252 for others to try.

Thanks for your help. I couldn't have done it without your valuable input.

Regards

Robert

Robomaticii, 2022-02-02:
Oh! and there it is. Guess you can see my user name is "chimney". Forgot to replace it with a generic in one step.

Hope the above helps.

Robert AKA Robomaticii

Gasman2014, 2022-02-02:
Je vous en prie 😉

John Pateman

themaninthesuitcase, 2022-02-02:
> Last step is to sign the KiCad.app. 
codesign -fs my-certificate /Applications/Kicad/KiCad.app 

This was the missing secret sauce for me.

yaqwsx, 2022-02-09:
Hi! I see that you made a progress with this issue. Would you mind writing a step-by-step guide and submitting it as a pull request to the documentation? 

PS: Is there anything we could ask from the KiCAD developers to make the whole process easier? I find the current situation on Mac really bad, however, since I am not a Mac use I am not sure if I fully understand what needs to change.

Robomaticii, 2022-02-09:
Jan

I will put together a step-by-step guide and submit it as a pull request.

However, I wish I knew exactly why it works. I am more of a hardware guy
and an artificer (someone who just makes things work) -  not a programmer.
I suspect it has something to do with how Mac OS handles certificates. I am
going to do some research to see if I can figure it out.

I will be in touch soon.

Robert

sslupsky, 2022-02-22:
I think it Is possible to solve this issue using an virtual environment  I did some investigation and it is possible to install the KiKit backend into a virtual environment that is created from the KiCad python.  For example:
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m venv /Users/stevenslupsky/Documents/kicad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit 
```

This installed a virtual environment in the KiKit plugin directory.  I can now open a terminal and activate that python environment and run KiKit:
```
source /Users/stevenslupsky/Documents/kicad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit/bin/activate
```

And I can confirm that KiKit is running properly:
```
(com_github_yaqwsx_kikit) barbie-en0:~ stevenslupsky$ which python3
/Users/stevenslupsky/Documents/kicad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit/bin/python3
(com_github_yaqwsx_kikit) barbie-en0:~ stevenslupsky$ which kikit
/Users/stevenslupsky/Documents/kicad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit/bin/kikit
(com_github_yaqwsx_kikit) barbie-en0:~ stevenslupsky$ kikit --version
kikit, version 1.0.2
```

The only remaining issue is to get the plugin to recognize the virtual environment.  I did this by adding the path to the site directory in the plugin (kikit/plugins/__init__.py):
(https://github.com/sslupsky/KiKit/blob/0e9d539789b57712be53f4cbf2a91552bea0bbff/pcm/kikit/plugins/__init__.py#L41-L47)

And this appears to work without having to code sign the KiCad app in macOS.  Of course, this is specific to my particular installation so the proposed solution here requires a little more work to extract the proper path from the virtual environment and add it to the site package search list.

sslupsky, 2022-02-22:
There is another way to accomplish the same thing without the change described above.  Create a file "kicad.pth" with the following:
```
# kicad 3rdparty package configuration

/Users/stevenslupsky/Documents/kicad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit/lib/python3.8/site-packages
```

Use the path to the directory you created the virtual environment you installed KiKit to.

Place the file in the following directory:
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages
```

Viola.

conradopoole, 2022-03-10:
> There is another way to accomplish the same thing without the change described above. Create a file "kicad.pth" with the following:
>
> ```
> # kicad 3rdparty package configuration [...]

I repeated your steps, both with the fix for init.py or the kicad.pth file, and in my case it still does not work unless I resign the Kicad.app .  It does install kikit in the virtual environment but the minute the app tries to load dependencies that load SO objects that were not in the original signature, it fails. Same issue as before, only that it now complains about paths in the virtual environment. An easy easy way to reproduce it is to launch the KyPython shell and run

` from kikit.plugin import importAllPlugins `

It fails with the same error as always due to signatures... See below...

[image]

Did you make any changes to your Notarization settings?

sslupsky, 2022-03-10:
@conradopoole What machine are you running?  I found things a little different if you are running on Apple Silicon.

The root cause is the numpy package.  The numpy package has a few binary libraries that cause the problem as a result of gatekeeper Library Validation.  Library Validation fails for any binaries that are not signed with the same Team ID unless Library Validation is disabled.  This is strictly enforced on Apple Silicon macs but apparently not on Intel macs.  This is why I noticed a difference in behaviour between the two architectures.

Regarding recreating the problem it is even simpler to do, just attempt to import numpy.

The solution to this problem requires that @yaqwsx include with KiKit the packages used by KiKit that are not included with the KiCad app bundle site-packages.  So at a minimum, @yaqwsx will need to include numpy because numpy is not included in the KiCad python site-packages.  Then, the KiKit plugin package, including the numpy package used by KiKit, needs to be signed by KiCad.  Thus, when the package is installed, it will pass library validation since both the plugin libraries and the python binaries are signed by KiCad.  I posted an issue on the KiCad gitlab repo and briefly discussed it with the KiCad team.  I think the KiCad developers are likely reluctant to disable library validation.  So, I think signing the plugin is the most appropriate solution.

It is possible to work around this for the moment but it requires the manual steps generally described here.  So, regarding your specific issue, are you running on Intel or Apple Silicon?  I have it working on both.  If running on Apple Silicon, you do need to re-sign the python binaries and the numpy libraries using your own certificate.  

I also found that the KiCad app bundle python creates cache files inside the app bundle.  From the work I did, I found that using PYTHONPATH and PYTHONPYCACHEPREFIX environment settings configured the python environment properly so python could find the site-packages directory for kikit and not pollute the KiCad app bundle with the cache files.  I installed kikit and the kikit dependencies in my own directory outside the KiCad app bundle.

I will see if I can put together a step by step set of instructions of how to do all of this.

sslupsky, 2022-03-10:
Here is the script I use to sign the proper items in the KiCad app bundle:
```
codesign --verbose=4 --force --sign "<put your certificate here>" /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Resources/Python.app
codesign --verbose=4 --force --sign "<put your certificate here>" /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3.8
codesign --verbose=4 --force --sign "<put your certificate here>" /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Python
codesign --verbose=4 --force --sign "<put your certificate here>" /Applications/KiCad/KiCad.app
```

I set the following environment variables in the "Preferences->Configure Paths" dialog:
```
PYTHONPATH /Users/stevenslupsky/Documents/KiCad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit/lib/python3.8/site-packages
PYTHONPYCACHEPREFIX /Users/stevenslupsky/Documents/KiCad/6.0/3rdparty/cache
PYTHONUSERBASE /Users/stevenslupsky/Documents/KiCad/6.0/3rdparty/plugins/com_github_yaqwsx_kikit
```

Since the numpy libraries are signed ad-hoc they will run if the python binaries are signed by you.  I don't think the PYTHONUSERBASE setting is required since I added the PYTHONPATH setting.

secolive, 2022-03-15:
Hi there, I have been experiencing the same difficulties while trying to install another tool (pcbdraw), which also uses numpy. Following the advice in this thread, I came up with the following plan (for reference).
1. create a new virtualenv based on KiCad's python
2. "link" KiCad's python to this new virtualenv (technically: include the virtualenv's site-packages into KiCad's)
3. resign KiCad
4. install whatever is needed in this virtualenv

I created the following script to perform steps 1-3:
`#!/bin/zsh

set -e -u

SCRIPT_DIR="$( dirname "$0" )"
SCRIPT_DIR="$( cd "$SCRIPT_DIR" && /bin/pwd )"

head1() {
  printf "\n"
  printf "%s\n" "$*" | sed 'h;s/./=/g;H;G;' 1>&2
}
notice() {
  printf "%s\n" "$*" 1>&2
}

V=3.8
KICAD_PYTHON="/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/$V"
CERTIFICATE="user@hostname CodeSigning"
PYENV_DIR="$SCRIPT_DIR/PyEnv"

head1 "Parameters"
echo "Kicad Python dir: $KICAD_PYTHON" 1>&2
echo "Python version: $V" 1>&2
echo "VirtualEnv dir: $PYENV_DIR" 1>&2
echo "Signinf certificate: $CERTIFICATE" 1>&2

# Create new virtualenv out of KiCad's python
head1 "Creation of new virtualenv"

notice "+ Create virtualenv"
"$KICAD_PYTHON/bin/python3" -m venv "$PYENV_DIR"

notice "+ Add KiCad's site-packages to virtualenv"
ls -d1 "$KICAD_PYTHON/lib/python$V/site-packages" > "$PYENV_DIR/lib/python$V/site-packages/KiCad.pth"

head1 "Update of KiCad"
notice "+ Add virtualenv's site-packages to KiCad's"
ls -d1 "$PYENV_DIR/lib/python$V/site-packages/" > "$KICAD_PYTHON/lib/python$V/site-packages/KiCad_plugins.pth"

notice "+ Resign KiCad"
codesign -fs "$CERTIFICATE" "$KICAD_PYTHON/Resources/Python.app"
codesign -fs "$CERTIFICATE" "/Applications/KiCad/KiCad.app"`

One of the tricks is that each python environment references the other's site-packages to make things easier.

Now I can use this virtualenv to install additional software:
`. PyEnv/bin/activate
pip3 install -U pip pillow numpy pcbdraw kikit`

And running python I can import everything successfully:
`$ python3
Python 3.8.2 (v3.8.2:7b3ab5921f, Feb 24 2020, 17:52:18)
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pcbnew
>>> import numpy
>>>`

It remains to be seen if this strategy works in the medium run (e.g. facing KiCad updates).

yaqwsx, 2022-03-15:
Just for reference - the related KiCAD issue: https://gitlab.com/kicad/code/kicad/-/issues/10943

sslupsky, 2022-03-15:
@secolive I believe you are missing one of the python binaries:
```
codesign --verbose=4 --force --sign "<put your certificate here>" /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3.8
```

Using an virtual environment is the proper way to ensure you can run kikit from the command line properly.  One thing to remember though is that the python bundled with KiCad does not use the virtual environment when run from within KiCad.  So I recommend using the environment variables I describe in my earlier post.  These variables ensure that when the **kikit plugin** is run from pcbnew, it has the correct environment configuration.

Fortunately, using the virtual environment will ensure the site-packages are retained across KiCad updates.  However, in order to use the site packages not included with KiCad, the KiCad app and python binaries have to be resigned each time KiCad is updated.

sslupsky, 2022-03-22:
@yaqwsx I think it Is close.  The user will have to set a couple environment variables in their Kicad configuration.

[11 more replies]

closed 2022-08-17 by yaqwsx