yaqwsx/KiKit #258: Fabrication generation is broken

marcovannoord, 2021-12-28:
**Describe the bug**: After upgrading my KiCad to v6.0.0 on Windows 10, and using the latest commit of the KiKit repo, i am unable to generate fabrication files for an existing board.

**Expected behavior**: i would expect the gerbers, `pos`, `bom` files to be present, but only the gerbers seem to be generated correctly

# To Reproduce

Source board: I'm afraid i cannot share the kicad_pcb file here
KiKit command:
```
kikit fab pcbway --assembly --no-drc --schematic cg_sensorboard.sch cg_sensorboard.kicad_pcb Assembly
``` 
or using `jlcpcb` instead of pcbway, does the same thing.

Perhaps relevant: this PCB has been made using KiCad 5 originally, but has been updated to the KiCad 6 file format. Also if i remove the `--no-drc` i will get another error, for which i will create a separate git issue

# Screenshots

```
kikit fab pcbway --assembly --no-drc --schematic cg_sensorboard.sch cg_sensorboard.kicad_pcb Assembly
Traceback (most recent call last):
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\Scripts\kikit-script.py", line 33, in <module>
    sys.exit(load_entry_point('KiKit===0.99-pre1.0-88.g674f43e', 'console_scripts', 'kikit')())
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\fab_ui.py", line 62, in pcbway
    return pcbway.exportPcbway(**kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\fab\pcbway.py", line 148, in exportPcbway
    components = extractComponents(schematic)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 149, in extractComponents
    symbols, instances = collectSymbols(filename)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\eeschema_v6.py", line 114, in collectSymbols
    sheetSExpr = parseSexprF(f)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 162, in parseSexprF
    s = readSexpr(stream)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 136, in readSexpr
    stream.shift("(")
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\sexpr.py", line 32, in shift
    raise ParseError(f"Expected '{expected}', got {repr(c)}")
TypeError: exceptions must derive from BaseException
```

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`  : `kikit, version 1.0.0+3.ga5ea593`
- source of installation (PyPI or upstream version): installed by running `pip install git+https://github.com/yaqwsx/KiKit@master` in my kicad command prompt
- your operating system Windows 10 19043

yaqwsx, 2021-12-28:
The problem is that the supplied schematics are in v5 format, however, KiKit is run with KiCAD 6 so it assumes the schematics are also in v6 format. Unfortunately, as there is no API for schematics so I parse the files manually, therefore, I cannot convert them automatically. Please, convert your schematics to v6 or run KiKit with KiCAD v5 on your v5 schematics.

I will implement automatic detection of the file format so I can at least show a usable error message.

marcovannoord, 2021-12-28:
Thanks for the quick reply! I see that the issue was between the chair and keyboard, i did not notice that the `.sch` got renamed to `kicad_sch`. Running kikit using the v6 schematic argument was the solution, and exporting to JLCPC now works again. 

Although i do not use them, but it might be useful to know: the pcbway export is currently broken.
```
kikit fab pcbway --assembly --no-drc --schematic cg_sensorboard.kicad_sch cg_sensorboard.kicad_pcb Assembly
Traceback (most recent call last):
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\KiCad\6.0\bin\Lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\Scripts\kikit.exe\__main__.py", line 7, in <module>
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1128, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1053, in main
    rv = self.invoke(ctx)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1659, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\click\core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\fab_ui.py", line 62, in pcbway
    return pcbway.exportPcbway(**kwargs)
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\fab\pcbway.py", line 157, in exportPcbway
    bom = collectBom(components, manufacturerFields, partNumberFields,
  File "C:\Users\Marco\Documents\KiCad\6.0\3rdparty\Python39\site-packages\kikit\fab\pcbway.py", line 45, in collectBom
    if c["unit"] != 1:
TypeError: 'Symbol' object is not subscriptable
```

yaqwsx, 2021-12-28:
It's not entirely your fault, KiKit should provide a better error message. Thanks for the report about PcbWay, I will fix this ASAP.

yaqwsx, 2021-12-28:
The problem with wrong version of schematics file was addressed in 596ea1622afcd03309550ec4d3a812cb4daf0a43, PcbWay fixed in 892e4b7a69312df4e44fd96b7d9a783119f59cc1.

closed 2021-12-28 by yaqwsx