yaqwsx/KiKit #260: KiCad process doesn't exit after plugin installation

qu1ck, 2021-12-29:
After installing KiKit through PCM and following additional install instructions for windows I got it working. But after closing the pcbnew and kicad windows the process still lingers. Usually it's because some dialogs are not destroyed correctly but I couldn't find anything suspicious. It also happens if KiKit plugin is not invoked at all which means the issue is in import code, not plugin code.

I suspect it's because of this

https://github.com/yaqwsx/KiKit/blob/master/kikit/plugin/__init__.py#L17

yaqwsx, 2021-12-29:
The line is actually needed for `StandardPaths` to work correctly. Is there a way I could detect if the module was imported within KiKit and not a standalone application?

qu1ck, 2021-12-29:
Then create the app when you need it, for example in `enable()`. That way it won't be created when plugin just imports the module.

Also I don't understand what's the point of `importAllPlugins()` ? Plugin instantiation and importantly `register()` call happens in generated kikit_plugin.py anyway. Why not move this to plugin files and get rid of the `kikit-plugins enable --all` step?

yaqwsx, 2021-12-29:
`kikit-plugins enable` is there for two reasons:
- v5 support where the plugins cannot be enabled via PCM
- for my testing scripts where I can quickly setup/teardown given configuration without the need to click anywhere. 

I will move the `PyApp` so it is not created globally.

qu1ck, 2021-12-29:
Ok, but `importAllPlugins()` still does exactly nothing as far as I can tell.

Your use cases for `kikit-plugins enable` are fine but forcing users to do it when they install from PCM is unnecessary imo.

yaqwsx, 2021-12-29:
`importAllPlugins()` imports all plugins defined in `availablePlugins` - the single source of truth holding all available plugins. Each plugin on import registers and ActionPlugin.

What do you mean by "forcing users to do it when they install from PCM is unnecessary IMO."? The users who installed KiKit via PCM are actually not supposed to use `kikit-plugins`. PCM is the substitution for that. `kikit-plugins` is actually no longer referenced from the documentation. Or what do I miss?

qu1ck, 2021-12-30:
> importAllPlugins() imports all plugins defined in availablePlugins - the single source of truth holding all available plugins. Each plugin on import registers and ActionPlugin.

Except importing is not enough for plugins to work in pcbnew, you have to instantiate them and call `register()` which `importAllPlugins()` doesn't do.

> What do you mean by "forcing users to do it when they install from PCM is unnecessary IMO."?

I mean that I had to do it to make plugins appear in pcbnew because of what I wrote above.

yaqwsx, 2021-12-30:
> importAllPlugins() doesn't do.
Currently, it does. A side-effect of loading the plugin module is registering it. It is messy (and I plan to refactor it), but it has historical architecture reasons and I didn't want to change it before the v1 release.

> I mean that I had to do it to make plugins appear in pcbnew because of what I wrote above.
I miss something, invoking `kikit-plugins` is not needed and it is actually discouraged (there is no note about it in the documentation - if there is, it is a bug). The installation procedure should be:
- install KiKit via pip (to fetch all the dependencies),
- install PCM package to show the icons in GUI (so the users don't have to copy any files anywhere).

yaqwsx, 2021-12-30:
d3edd5e should solve this. Upgrading via pip to the upstream version should be enough. I am sorry for the misunderstanding, I wasn't properly focused or something and I missed it /o\.

qu1ck, 2021-12-30:
No worries, thanks for the fix.

I tested it and both issues are resolved, no more manual enabling of plugins needed and kicad process terminates correctly. I recommend you push this to pypi soon.

closed 2021-12-30 by qu1ck

yaqwsx, 2021-12-30:
Thanks, I will fix some dependency issues and release v1.0.1; hopefully today.