yaqwsx/KiKit #263: Cannot perform: unsupported operand types

Bouni, 2021-12-30:
**Describe the bug**: When I click "Panelize" I get an error message

[image]

**Expected behavior**: Expcted a panel out of my single board

# To Reproduce

# Screenshots

[image]

# Version

Specify which version of KiKit do you use:
- 1.0.0
- PCM + pip
- Windows 10

yaqwsx, 2022-01-02:
I just tried to reproduce this problem with some of my boards without success. Could you share more precise procedure to reproduce and to share the board (if you cannot publish it, you can share it with me privately).

Bouni, 2022-01-03:
Here is the kicad_pcb file:
[ESP32CAM-POWER.zip](https://github.com/yaqwsx/KiKit/files/7801691/ESP32CAM-POWER.zip)

yaqwsx, 2022-01-03:
What version of PcbNew was used to create this file? I get:
[image]

Using `Version: 6.99.0-a9ca49a600~144` which is the most-recent nightly.

Bouni, 2022-01-03:
I used a nightly build from last week and updated today to the latest nightly but cannot say for sure if I posted the kicad_pcb file before or after the update 😅

Bouni, 2022-01-03:
At the moment I run `kicad-msvc.r25922.211820a689-x86_64.exe` downloaded from here: https://kicad-downloads.s3.cern.ch/windows/nightly/kicad-msvc.r25922.211820a689-x86_64.exe

That gives me this versioninfo in PCB Editor:

```
Application: KiCad PCB Editor (64-bit)

Version: (6.99.0-224-g211820a689), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jan  3 2022 09:00:30
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON

```

yaqwsx, 2022-01-04:
Ah, the nighly is out. There is a number of breaking changes (and possibly Python API regression) coming. This one is that KiCAD stopped using floats for angles and instead, it provides `EDA_ANGLE`. This is what you observe.  Based on the experiences with 5.99, I am not going to support 6.99 for a while. I guess the support could be introduced in August/September 2022 or if the new, stable API appears.

closed 2022-01-04 by yaqwsx