yaqwsx/KiKit #265: If I perform a panelization, the panel.kicad_pcb file is not generated but the .kicad_pcb file of the project is modified.

tistructor, 2021-12-31:
**Describe the bug**: A clear and concise description of what the bug is.

I was doing a paneling test of a small board.
I set the parameters and when I press panelize the panel.kicad_pcb file is not generated but the .kicad_pcb file of the project is modified.

**Expected behavior**: A clear and concise description of what you expected to happen.

# To Reproduce

Source board: <include your `.kicad_pcb` file here>
[X_Y_Z_Limiter_V1_3.zip](https://github.com/yaqwsx/KiKit/files/7796601/X_Y_Z_Limiter_V1_3.zip)

KiKit command:
```kikit panelize \
    --layout 'hspace: 2mm; vspace: 2mm; cols: 6' \
    --tabs 'type: fixed' \
    --cuts 'type: mousebites; spacing: 1mm; offset: 0.2mm; prolong: 0.5mm' \
    --post 'millradius: 1mm' \
    D:\RYZEN\DATI\Progetti\CNC-Laser\X_Y_Z_Limiter_V1_3\X_Y_Z_Limiter_V1_3.kicad_pcb panel.kicad_pcb

{
    "layout": {
        "hspace": "2mm",
        "vspace": "2mm",
        "cols": "6"
    },
    "tabs": {
        "type": "fixed"
    },
    "cuts": {
        "type": "mousebites",
        "spacing": "1mm",
        "offset": "0.2mm",
        "prolong": "0.5mm"
    },
    "post": {
        "millradius": "1mm"
    }
}

<include the KiKit invocation here>
```

# Screenshots
[image]

https://user-images.githubusercontent.com/24445373/147832334-89fbcb8b-448e-4840-a771-c2aaed35d1dd.mp4

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version` kikit, version 1.0.0
- source of installation (PyPI or upstream version)
- your operating system windows10 Pro

yaqwsx, 2021-12-31:
This is expected, but at the moment undocumented,, bahavior. In GUI you select what should be panelized, but the result is placed into the currently opened document of Pcbnew. You are actually expected to run KiKit panelized from a standalone Pcbnew instance. The `panel.kicad_pcb` is just a placeholder if you use the GUI to construct the KiKit command. I should replace it with something that hints it is only a placeholder, possibly `<your_outputfile>`.

The GUI should be better documenterd, I will improve the documentation in this regard.

tistructor, 2021-12-31:
Thank you,
I create a standalone file for tiling.
I did some command line tests and the operation was successful even though I received an alert window from wxWidgets but without causing any problems.

https://user-images.githubusercontent.com/24445373/147833379-3c713344-beb3-47d1-b0da-a883d2f080fd.mp4

Last thing, isn't it better to create the panel.kicad_pcb file instead of modifying the project file?

Thanks and congratulations for the plugin.

yaqwsx, 2021-12-31:
I don't think it wise to create the panel file. The main use-case for the GUI is to create the command a quickly tweak the panel. For that you want to see what was created as fast as possible. When we would create a new file, you would have to open a new window od PcbNew. That gets quickly annoying.

Once you have the desired command, you should put it inside your Makefile and generate panels automatically.

yaqwsx, 2022-01-03:
The documentation was improved in eed67f79859e6f0c0e8b06dc3fbdf5e221add4f3

closed 2022-01-03 by yaqwsx