yaqwsx/KiKit #266: Some missing tabs when using alternation: cols

mairas, 2021-12-31:
**Describe the bug**: I am trying to panelize a board for JLCPCB fabbing with `alternation: cols`. For some reason, when using that option, some mousebite tabs are missing (see the screenshot).

**Expected behavior**: Without `alternation: cols`, tabs are fine.

I'd expect the tabs to be identical in each case.

# To Reproduce

Source board: https://github.com/mairas/SH-ESP32-engine-hat/tree/release_1.0.0 (note the `release_1.0.0` branch)
KiKit command:
```
kikit panelize \
   --layout 'grid; space: 3mm; rows: 2; cols: 2' \
   --tabs 'fixed; width: 3mm' \
   --cuts 'mousebites; drill: 0.5mm; spacing: 1mm; prolong: 0.5mm' \
   --framing 'frame; width: 5mm; space: 3mm; cuts: both' \
   --source 'tolerance: 10mm' \
   --post 'millradius: 1mm' \
   SH-ESP32-engine-hat.kicad_pcb SH-ESP32-engine-hat-panel.kicad_pcb
```

# Screenshots

[image]

[image]

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`

```
$ kikit --version
kikit, version 1.0.0
```

- source of installation (PyPI or upstream version)

`pip3 install kikit`

- your operating system

Ubuntu 20.04.3 LTS (aarch64)

mairas, 2021-12-31:
Oh, and have a happy and a safe new year 2022!

yaqwsx, 2022-01-02:
Thanks for the report. I confirm I can reproduce it. Current progress is that I have traced the problem to wrongly generated partition lines between the boards:

[image]
[image]

yaqwsx, 2022-01-03:
Oh, got it in 48d7dea1b18a2f69bffb05ddca25cc2d986d868f. You can manually upgrade to upstream or wait for the v1.0.1 release (I will make it today or tomorrow).

When I was playing with timing down tolerance I didn't notice I accidentally affected the filter for partition line seeds. 

I have to come up with how to automatically test for these kinds of regressions so it doesn't happen in the future.

closed 2022-01-03 by yaqwsx