yaqwsx/KiKit #267: Feature request: Run ERC on schematic

korken89, 2022-01-02:
Hi and thanks for a great tool!

I was playing around with creating a CI job using KiKit, but I found one thing I was missing: to run ERC on the schematic.
The DRC exists and works great, I'd like to put in a feature request for the same functionality on schematics.

Thanks!

yaqwsx, 2022-01-02:
Unfortunately, this is not currently possible in KiCAD as there is no API. We are promised to have something in v7.

If you are only interested in CI, then look at https://github.com/INTI-CMNB/KiBot - that is a tool for CI that should be able to do this (it launches KiCAD in virtual screen and send keystrokes to it). We do not plan to implement anything like this into KiCAD as it is hard to make it multiplatform and actually work in common user interface

korken89, 2022-01-03:
Thanks for the update!
Unfortunate that is didn't have an API yet

yaqwsx, 2022-09-15:
Unfortunately, it is pretty sure that the schematic API won't make it to KiCAD 7. So we have to postpone this feature.

yaqwsx, 2024-03-10:
With KiCAD 8, there is an option to run `kicad-cli sch erc` and `kicad-cli pcb drc`. I am unsure if adding `kikit erc` makes any sense. I think KiKit cannot provide more value than the built-in commands.

korken89, 2024-03-11:
I think it's not needed.

closed 2024-03-11 by korken89