yaqwsx/KiKit #270: Clearance is always off for 0.005 mm

yaqwsx, 2022-01-03:
### Discussed in https://github.com/yaqwsx/KiKit/discussions/269

<div type='discussions-op-text'>

<sup>Originally posted by **joric** January  3, 2022</sup>
Each time i'm running drc I'm getting tolerance drc errors. I have board clearance set to 0.2 mm but kikit somehow rownds down and ends with 1.995 everywhere and DRC fails. Anyone else faced that, how to fix?

See https://github.com/joric/nrfmicro/runs/4684323269

```
** Found 6 DRC violations: **
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(32.4481 mm, 28.4525 mm): Track [DATA+] on B.Cu, length 4.9642 mm
  @(32.8476 mm, 28.2420 mm): Track [POWER_PIN] on B.Cu, length 5.0085 mm
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(30.9276 mm, 26.9320 mm): Track [DATA+] on B.Cu, length 2.1503 mm
  @(32.8476 mm, 28.2420 mm): Track [POWER_PIN] on B.Cu, length 5.0085 mm
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(28.5721 mm, 25.9375 mm): Track [DATA-] on F.Cu, length 0.7010 mm
  @(28.9716 mm, 25.2716 mm): Track [CC1] on F.Cu, length 0.3875 mm
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(28.5721 mm, 25.2365 mm): Track [DATA-] on F.Cu, length 0.1421 mm
  @(28.9716 mm, 24.6000 mm): Track [CC1] on F.Cu, length 0.6716 mm
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(32.8476 mm, 33.2505 mm): Track [POWER_PIN] on B.Cu, length 0.9892 mm
  @(32.4481 mm, 28.4525 mm): Track [DATA+] on B.Cu, length 4.9642 mm
[clearance]: Clearance violation (netclass 'Default' clearance 0.2000 mm; actual 0.1995 mm) Severity: error
  Rule: netclass 'Default'
  @(28.9716 mm, 24.6000 mm): Track [CC1] on F.Cu, length 0.6716 mm
  @(28.5721 mm, 25.9375 mm): Track [DATA-] on F.Cu, length 0.7010 mm
DRC failed. See report above. No files produced
```</div>

This seems like a problem on the KiCAD side, we should create a minimal counter-example.

yaqwsx, 2022-01-03:
This seems to be a bug in KiCAD. Related KiCAD issue: https://gitlab.com/kicad/code/kicad/-/issues/10221

FrankvVeelen, 2022-01-05:
Is there a way to not run DRC or ignore DRC errors for now, as a workaround?

yaqwsx, 2022-01-07:
The bug was fixed in KiCAD and the bugfix will be part of the 6.0.1 release that should be available since January 15, 2022.

crides, 2022-01-08:
I already don't see this problem in Kicad, but Kikit still generate these DRC errors

```
Application: KiCad PCB Editor

Version: 6.0.0, release build

Libraries:
	wxWidgets 3.0.5
	libcurl/7.80.0 OpenSSL/1.1.1m zlib/1.2.11 brotli/1.0.9 zstd/1.5.1 libidn2/2.3.2 libpsl/0.21.1 (+libidn2/2.3.0) libssh2/1.10.0 nghttp2/1.46.0

Platform: Linux 5.14.16-arch1-1 x86_64, 64 bit, Little endian, wxGTK, awesome, x11

Build Info:
	Date: Dec 30 2021 00:19:57
	wxWidgets: 3.0.5 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.78.0
	OCC: 7.5.3
	Curl: 7.80.0
	ngspice: 35
	Compiler: GCC 11.1.0 with C++ ABI 1016

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```

yaqwsx, 2022-01-08:
@crides: Please, read gitlab.com/kicad/code/kicad/-/issues/10221

It is a bug in KiCAD that returns different DRC results when the DRC is invoked from API and from GUI. The fixed KiCAD will be probably released on January 15, 2022.

crides, 2022-01-08:
Oops, sorry I ignored the link

yaqwsx, 2022-01-18:
Unfortunately, there are more things wrong so the fix in 6.0.1 is not sufficient. See the GitLab issue for more details.

yaqwsx, 2022-01-29:
There has been progress in KiCAD, therefore if you compile the current KiCAD v6 branch, it should work.

putyn, 2022-02-25:
I guess the only way to overcome this, until fixed, is to not run DRC when exporting gerbers, right?

yaqwsx, 2022-02-25:
Yes. You could also compile custom version of KiCAD that includes the patch.

putyn, 2022-02-25:
I am mainly working on Windows, I wouldn't dare try that :)

closed 2022-03-17 by yaqwsx

duckyb, 2022-05-17:
Is there a kikit docker image which includes this fix?

yaqwsx, 2022-05-18:
I updated the `yaqwsx/kikit:latest` image to contain the new KiCAD.