yaqwsx/KiKit #272: V6 Windows install Fails when Building Regex

BillyPocketNC, 2022-01-03:
Kit Kit Not installing for Windows KiCad V6.0 Build problems
install fails when building wheel for regex

'''
>pip3 install kikit
Defaulting to user installation because normal site-packages is not writeable
Collecting kikit
  Using cached KiKit-1.0.1-py3-none-any.whl (206 kB)
Collecting click>=7.1
  Using cached click-8.0.3-py3-none-any.whl (97 kB)
Collecting solidpython>=1.1
  Using cached solidpython-1.1.1-py3-none-any.whl (86 kB)
Collecting shapely>=1.7
  Using cached Shapely-1.8.0-cp39-cp39-win_amd64.whl (1.1 MB)
Collecting pybars3>=0.9
  Using cached pybars3-0.9.7-py3-none-any.whl
Collecting markdown2>=2.4
  Using cached markdown2-2.4.2-py2.py3-none-any.whl (34 kB)
Requirement already satisfied: numpy>=1.21 in c:\program files\kicad\6.0\bin\lib\site-packages (from kikit) (1.21.5)
Collecting commentjson>=0.9
  Using cached commentjson-0.9.0-py3-none-any.whl
Collecting pcbnewTransition==0.2.0
  Using cached pcbnewTransition-0.2.0-py3-none-any.whl (4.3 kB)
Collecting colorama
  Using cached colorama-0.4.4-py2.py3-none-any.whl (16 kB)
Collecting lark-parser<0.8.0,>=0.7.1
  Using cached lark_parser-0.7.8-py2.py3-none-any.whl
Collecting PyMeta3>=0.5.1
  Using cached PyMeta3-0.5.1-py3-none-any.whl
Collecting euclid3<0.2.0,>=0.1.0
  Using cached euclid3-0.1-py3-none-any.whl
Collecting pypng<0.0.20,>=0.0.19
  Using cached pypng-0.0.19-py3-none-any.whl
Collecting PrettyTable==0.7.2
  Using cached prettytable-0.7.2-py3-none-any.whl
Collecting regex<2020.0,>=2019.4
  Using cached regex-2019.12.20.tar.gz (679 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: regex
  Building wheel for regex (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: 'c:\program files\kicad\6.0\bin\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\{USER}\\AppData\\Local\\Temp\\pip-install-2pnv_uvz\\regex_a248fb9967d94d43a25a90bc31630013\\setup.py'"'"'; __file__='"'"'C:\\Users\\{USER}\\AppData\\Local\\Temp\\pip-install-2pnv_uvz\\regex_a248fb9967d94d43a25a90bc31630013\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d 'C:\Users\{USER}\AppData\Local\Temp\pip-wheel-w0vpcvw4'
       cwd: C:\Users\{USER}\AppData\Local\Temp\pip-install-2pnv_uvz\regex_a248fb9967d94d43a25a90bc31630013\
  Complete output (19 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-3.9
  creating build\lib.win-amd64-3.9\regex
  copying regex_3\__init__.py -> build\lib.win-amd64-3.9\regex
  copying regex_3\regex.py -> build\lib.win-amd64-3.9\regex
  copying regex_3\_regex_core.py -> build\lib.win-amd64-3.9\regex
  copying regex_3\test_regex.py -> build\lib.win-amd64-3.9\regex
  running build_ext
  building 'regex._regex' extension
  creating build\temp.win-amd64-3.9
  creating build\temp.win-amd64-3.9\Release
  creating build\temp.win-amd64-3.9\Release\regex_3
  C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Program Files\KiCad\6.0\bin\include -IC:\Program Files\KiCad\6.0\bin\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcregex_3\_regex.c /Fobuild\temp.win-amd64-3.9\Release\regex_3\_regex.obj
  _regex.c
  regex_3\_regex.c(48): fatal error C1083: Cannot open include file: 'Python.h': No such file or directory
  error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
  ----------------------------------------
  ERROR: Failed building wheel for regex
  Running setup.py clean for regex
Failed to build regex
Installing collected packages: regex, pypng, PyMeta3, PrettyTable, lark-parser, euclid3, colorama, solidpython, shapely, pybars3, pcbnewTransition, markdown2, commentjson, click, kikit
  Attempting uninstall: regex
    Found existing installation: regex 2021.11.10
[... 17 lines cut ...]
    running build_ext
    building 'regex._regex' extension
    creating build\temp.win-amd64-3.9
    creating build\temp.win-amd64-3.9\Release
    creating build\temp.win-amd64-3.9\Release\regex_3
    C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Program Files\KiCad\6.0\bin\include -IC:\Program Files\KiCad\6.0\bin\include -IC:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools\VC\Tools\MSVC\14.29.30133\include -IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\ucrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\shared -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\um -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\winrt -IC:\Program Files (x86)\Windows Kits\10\include\10.0.19041.0\cppwinrt /Tcregex_3\_regex.c /Fobuild\temp.win-amd64-3.9\Release\regex_3\_regex.obj
    _regex.c
    regex_3\_regex.c(48): fatal error C1083: Cannot open include file: 'Python.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\BuildTools\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2
    ----------------------------------------
  Rolling back uninstall of regex
  Moving to c:\users\{USER}\documents\kicad\6.0\3rdparty\python39\site-packages\regex-2021.11.10.dist-info\
   from C:\Users\{USER}\Documents\KICAD\6.0\3rdparty\Python39\site-packages\~egex-2021.11.10.dist-info
  Moving to c:\users\{USER}\documents\kicad\6.0\3rdparty\python39\site-packages\regex\
   from C:\Users\{USER}\Documents\KICAD\6.0\3rdparty\Python39\site-packages\~egex
ERROR: Command errored out with exit status 1: 'c:\program files\kicad\6.0\bin\python.exe' -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\{USER}\\AppData\\Local\\Temp\\pip-install-2pnv_uvz\\regex_a248fb9967d94d43a25a90bc31630013\\setup.py'"'"'; __file__='"'"'C:\\Users\\{USER}\\AppData\\Local\\Temp\\pip-install-2pnv_uvz\\regex_a248fb9967d94d43a25a90bc31630013\\setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record 'C:\Users\{USER}\AppData\Local\Temp\pip-record-c9dtwomw\install-record.txt' --single-version-externally-managed --user --prefix= --compile --install-headers 'C:\Users\{USER}\Documents\KiCad\6.0\3rdparty\Python39\Include\regex' Check the logs for full command output.'''

# To Reproduce

KiKit command:
```
pip install kitkit
or
pip3 install kitkit
```

yaqwsx, 2022-01-03:
Did you first install Solidpython manually as instructed in the [installation guide](https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md)?

```
pip install https://github.com/SolidCode/SolidPython/archive/refs/heads/master.zip
```

The older versions of this library carry a dependency that fails to build as you observe. There is no official release, so we have to install the upstream version of the library manually.

BillyPocketNC, 2022-01-03:
Thanks @yaqwsx Installing Solidpython first fixed it.

closed 2022-01-03 by BillyPocketNC