yaqwsx/KiKit #273: PCB outline using KiCad rectangle function fails to panelize

Obrekr, 2022-01-03:
**Describe the bug**: 
Trying to panelize a PCB created with KiCad 6.0 fails when drawing the outline with the `Rectangle` function.
The message generated when panellising is
```
An error occurred: Test.kicad_pcb: Discontinuous outline at [107.0, 69.0]
```

After converting the graphical rectangle to lines by right-clicking and selecting `Create from Selection`→`Create Lines from Selection` the PCB can be panellised.

**Expected behavior**: 
Using the rectangle function for PCB outlines in new KiCad 6.0 can be used to draw the outline without crashing KiKit panelization.
 
# To Reproduce

Source board: [Test.kicad_pcb.zip](https://github.com/yaqwsx/KiKit/files/7802918/Test.kicad_pcb.zip)
KiKit command:
```
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/kikit panelize --layout 'grid; rows: 2; cols: 1;' --tabs full --cuts vcuts Test.kicad_pcb panel.kicad_pcb 
```

# Version

Specify which version of KiKit do you use:
- `kikit --version`: kikit, version 1.0.1
- installed from pypi with `/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 -m pip install --upgrade pip numpy pillow kikit` and then code signing it manually with `codesign -fs "code-sign-ssr" /Applications/Kicad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/Resources/Python.app` after creating the certificate with Keychain.app
- running on macOS M1

yaqwsx, 2022-01-03:
This feature is currently missing (as there were no rectangles in v5); I will implement this soon.

yaqwsx, 2022-01-06:
Implemented in 090578d

closed 2022-01-06 by yaqwsx