yaqwsx/KiKit #275: An error occurred: Uknown type '<class 'shapely.geometry.collection.GeometryCollection'>' of substrate geometry

danielkucera, 2022-01-04:
**Describe the bug**:
Getting error when trying to run kikit like this:
```
danman@silverhorse:~/Projects/iot-meter/iec62056-optical$ cat panelize.sh 
kikit panelize --layout 'grid; rows: 4; cols: 4; space: 0mm' \
	--source 'tolerance: 15mm' \
	--tabs 'fixed; width: 5mm' \
	--cuts 'mousebites; drill: 0.5mm; spacing: 1mm; offset: -0.1mm' \
	--framing 'tightframe; width: 1.9mm; space: 0.1mm' \
	--fiducials '3fid; hoffset: 2.5mm; voffset: 2.5mm; coppersize: 2mm; opening: 1mm;' \
	iec62056-optical.kicad_pcb panel.kicad_pcb 
danman@silverhorse:~/Projects/iot-meter/iec62056-optical$ ./panelize.sh 
An error occurred: Uknown type '<class 'shapely.geometry.collection.GeometryCollection'>' of substrate geometry
No output files produced
```

**Expected behavior**: 
Create panel PCB.

# To Reproduce

Source board: https://raw.githubusercontent.com/danielkucera/iec62056-optical/d3b244f0d8d6a004dd5dfbfb0c4d36389db9ebc3/iec62056-optical.kicad_pcb

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`: kikit, version 1.0.2
- source of installation (PyPI or upstream version): pip
- your operating system: Ubuntu 
- KiCAD version: 6.0.0-d3dd2cf0fa~116~ubuntu21.10.1, release build

yaqwsx, 2022-01-04:
For reasons unknown to me the new version of Shapely, I started enforcing as a dependency, returns `GeometryCollection` instead of `PolygonCollection` in this case. Fixed in b30ccbd

closed 2022-01-04 by yaqwsx

danielkucera, 2022-01-04:
Thank you, I hope I am not bothering too much. :)

yaqwsx, 2022-01-04:
I am thankful for the reports! Testing KiKit properly is hard as everyone uses KiCAD differently, so there are uncaught cases.