yaqwsx/KiKit #276: Can't get kikit PCM side to work....

UnexpectedMaker, 2022-01-05:
I'm on a Mac - macOS 12.1

I have kikit installed via pip (installed today).
```
❯ kikit --help         
Usage: kikit [OPTIONS] COMMAND [ARGS]...

Options:
  --version  Show the version and exit.
  --help     Show this message and exit.

Commands:
  drc       Validate design rules of the board
  export    Export KiCAD boards
  fab       Export complete manufacturing data for given fabrication houses
  modify    Modify board items
  panelize  Panelize boards
  present   Prepare board presentation
  separate  Separate a single board out of a mu
```

Seems to be installed ok...   so I installed the PCM GUI stuff in KiCD 6.0 release, and no icon come sup - when I do a `refresh plugins` I get the following error:

[image]

I've been over the instructions again and again... it just wont work :(

Any ideas?

yaqwsx, 2022-01-05:
There are some problems on Mac - you probably didn't install it in the same Python interpreter as Pcbnew uses. There is another issue for that: #252. There are some proposals on how to fix it.

Unfortunately, I don't have access to Mac, so I cannot debug this by myself. Try reading #252 and please let me know if you have any luck resolving this.

UnexpectedMaker, 2022-01-05:
Yup, had gone through that after I posted... and am up to this...
```
 ❯ PYTHONPATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/ DYLD_LIBRARY_PATH=/Applications/KiCad/KiCad.app/Contents/Frameworks/ kikit panelize \
    --layout 'grid; rows: 2; cols: 2;' \
    --tabs full \
    --cuts vcuts \
    doc/resources/conn.kicad_pcb panel.kicad_pcb
An error occurred: dlopen(/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so, 0x0002): Library not loaded: @rpath/Versions/3.8/Python
  Referenced from: /Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/_pcbnew.so
  Reason: tried: '/Applications/KiCad/KiCad.app/Contents/Frameworks//Python' (no such file), '/usr/lib/Python' (no such file)
No output files produced
```

But doing the whole "install kikit with the cicada python version" seems like a nasty work around.

I don't have or use docker...  guess I have to go down that rabbit hole, or just hand panelist my boards... only have 3 unique complex panels to do ;)

Hopefully this will all be fixed soon, by someone smarter than me!

Cheers!

yaqwsx, 2022-01-05:
This is, unfortunately, caused by packaging KiCAD on Mac and I don't have control over this :/

yaqwsx, 2022-01-05:
PS: Docker is really easy to use. You just don't get the GUI.

UnexpectedMaker, 2022-01-05:
> This is, unfortunately, caused by packaging KiCAD on Mac and I don't have control over this :/

Oh, I know - wasn't "looking at you" for the fix ;) 

> PS: Docker is really easy to use. You just don't get the GUI.

Well, I'm about to find out!

closed 2022-01-05 by UnexpectedMaker