yaqwsx/KiKit #278: Chybí V-Cuty

MiroslavBurda, 2022-01-06:
**Describe the bug**: Při panelizaci chybí svislé V-Cuty

# To Reproduce

Source board: 
[Robotarna555.zip](https://github.com/yaqwsx/KiKit/files/7823957/Robotarna555.zip)

KiKit command:
```
kikit panelize \
    --layout 'rows: 2; cols: 3' \
    --cuts 'type: vcuts' \
    C:\KiCad\Robotarna555\Robotarna555.kicad_pcb panel.kicad_pcb
```

# Screenshots

If applicable, add screenshots to help explain your problem.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`
- source of installation (PyPI or upstream version)
- your operating system

yaqwsx, 2022-01-07:
Problém je v tom, že výchozí typ můstků je `spacing`, jenže tady se očividně snažíš vyrobit panel bez můstků (takže správný typ je `full`). Přemýšlím, jestli má KiKit při vzdálenosti desek 0 řvát, padat nebo tiše pokračovat. Anebo jestli je to jen problém dokumentace. Každopádně v tomto případě je třeba použít typ můsků `full`.

yaqwsx, 2022-01-08:
Zohledněno v bc2f13ae9763f16bb067610cd63abe5d9b93bf6a

closed 2022-01-08 by yaqwsx