yaqwsx/KiKit #279: Installed but not present

tormyvancool, 2022-01-06:
Please I described everything here
[https://forum.kicad.info/t/kitkit-plugin-issue-installed-but-not-present/32867/11](url)

it seems I'm not the only one to get this issue

KiCAD 6.0.0
Windows 11

# To Reproduce

Source board: <include your `.kicad_pcb` file here>
KiKit command:
```
<include the KiKit invocation here>
```

# Screenshots

here you find also the screenshots:
[https://forum.kicad.info/t/kitkit-plugin-issue-installed-but-not-present/32867/11](url)

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`
- source of installation (PyPI or upstream version)
- your operating system

dronecz, 2022-01-07:
Did you tried to upgrade KiKit? I had same issue and I just now used command `pip install -U kikit` to upgrade KiKit to latest version and window with error does not show up and I can see KiKit icons in KiCad 😉

tormyvancool, 2022-01-07:
> 
I followed your suggestion. It doesn't work as well.

yaqwsx, 2022-01-07:
Based on the screenshot you provided: [image]

I believe that you used "Windows Command Prompt" instead of "KiCAD Command Prompt". They are two different things - see them side-by-side:

[image]

If you really used the KiCAD Command prompt for the installation, then please verify that you can invoke `kikit --help` in it. If not, please send me the output of `where python` (everything inside KiCAD Command Prompt).

conradopoole, 2022-01-08:
Also the screenshot says kitkit too many Ts or you were thinking of a KitKat :-)

tormyvancool, 2022-01-08:
> 

This I corrected and the kikit --help worked as I said in the same topic if you scroll down to read it :-)

tormyvancool, 2022-01-08:
> Based on the screenshot you provided: [image]
> 
> I believe that you used "Windows Command Prompt" instead of "KiCAD Command Prompt". They are two different things - see them side-by-side:
> 
> [image]
> 
> If you really used the KiCAD Command prompt for the installation, then please verify that you can invoke `kikit --help` in it. If not, please send me the output of `where python` (everything inside KiCAD Command Prompt).
> 
> The problem with OK button was addressed in #262, however, I haven't updated the PCM package in the official repository (as this is the only change and I wasn't considering it as significant).

Thank you for the help. Yes I opened the wrong command line window. Now it works. thank you for the help

closed 2022-01-08 by yaqwsx

tormyvancool, 2022-01-16:
Pay attention.

I updated to 6.0.1 and the issue is back.
moreover clicking on the button OK nothing occurs (annoying)

now, back to search WHERE the kicad console is located :-D

yaqwsx, 2022-01-17:
Ah, you probably have to reinstall KiKit as it seems that the path with modules changes with every new version.

The KiCad Command Prompt is in the start menu, just type it and you will find it.

tormyvancool, 2022-01-17:
Thank you so much. Did it and it works now.
Ok, then I have to remember that to each update, this is a command I have to give