yaqwsx/KiKit #280: kikit panelize command "extractboard" missing

noanight, 2022-01-07:
**Describe the bug**:  
trying to run the example project RB0002-BatteryPack from https://cadlab.io/project/22740/master/files. it fails with 
Error: Got unexpected extra arguments (70 40 batteryPack/batteryPack.kicad_pcb build/charger.kicad_pcb)

**Expected behavior**: working example
# To Reproduce

Source board: <include your `.kicad_pcb` file here>
KiKit command:
```
make

kikit panelize extractboard -s 175 50 70 40 batteryPack/batteryPack.kicad_pcb build/charger.kicad_pcb
Usage: kikit panelize [OPTIONS] INPUT OUTPUT
Try 'kikit panelize --help' for help.

Error: Got unexpected extra arguments (70 40 batteryPack/batteryPack.kicad_pcb build/charger.kicad_pcb)
make: *** [Makefile:12: build/charger.kicad_pcb] Fehler 2

```
# Version
Specify which version of KiKit do you use:
- the output of `kikit --version` : kikit, version 1.0.2
- source of installation: PyPI 
- your operating system: Ubuntu 20.04

yaqwsx, 2022-01-07:
The example uses an old version (v0.7), the command was renamed to `separate`. It also has different CLI parameters. See https://github.com/yaqwsx/KiKit/blob/master/doc/multiboard.md

I should either upgrade the example or remove the reference to it.

closed 2022-01-07 by noanight

yaqwsx, 2022-01-07:
I removed the misleading example in 4e79fcb. There are plan for more thorough examples in the near future.