yaqwsx/KiKit #281: "No KiKit backend found!"

MarkJeronimus, 2022-01-07:
"No KiKit backend found! You probably installed KiKit only via PCM." -- Incorrect!

I installed KiKit via pip3, then installed it via PCM, and my KiCad 6.00 complains that it cannot find the backend. How would I go about debugging and fixing this?

Some information I can give up front:

```
$ python3 /home/mark/.local/bin/kikit --version
kikit, version 0.6.1
$ which kikit
/home/mark/.local/bin/kikit
```
Python2 is my default python (because my Ubuntu 18.04 LTS requires it be this way) and I have Python3 installed besides it.
I installed KiCad 6.00 via the official release PPA from their website (not the nightly or Canonical repository)

yaqwsx, 2022-01-07:
You have an ancient version of KiKit. The PCM package works only with versions 1 and newer. You have version 0.6.1. Please upgrade KiKit.

MarkJeronimus, 2022-01-09:
how do I upgrade? When I execute `pip3 install -U kikit` I get this version.

yaqwsx, 2022-01-09:
This is strange; the newest version is 1.0.2 (https://pypi.org/project/KiKit/), try `pip3 install -U kikit==1.0.2` to force a particular version.

MarkJeronimus, 2022-01-09:
```bash
$ pip3 install -U kikit==1.0.2
Collecting kikit==1.0.2
  Could not find a version that satisfies the requirement kikit==1.0.2 (from versions: 0.1.0, 0.1.1, 0.2.0, 0.2.1, 0.2.2, 0.2.3, 0.2.4, 0.3, 0.4, 0.5, 0.5.1, 0.5.2, 0.5.3, 0.6.0, 0.6.1)
No matching distribution found for kikit==1.0.2
```

yaqwsx, 2022-01-09:
This is strange; it seems like your pip cannot fetch the most recent version from pip. You should check your python installation. In meantime, you can try to install it directly from the release page (`pip install <nameofthefile>`).

MarkJeronimus, 2022-01-09:
I found something in the verbose output
```
 The package https://files.pythonhosted.org/packages/4c/8d/a3042cc67dbd01ed435801f330a94e477524c720869ce3db25b1ce035bbc/KiKit-1.0.2.tar.gz#sha256=3a01b8d7b8ee83f13ac2d7144538e6df6a65d5186f96237a7b274698383f1b62 (from https://pypi.org/simple/kikit/) 
(requires-python:>3.7) is incompatible with the pythonversion in use. Acceptable python versions are:>3.7
```

MarkJeronimus, 2022-01-09:
I managed to install Python 3.10 and KiKit 1.0.2

```
$ kikit --version
kikit, version 1.0.2
```

but after installing the frontend via the PCM I still get the same error. Could this be a path or virtualenv issue? (I have no experience with either)

yaqwsx, 2022-01-09:
You have to use the Python that KiCAD is compiled against. And I am afraid that on Ubuntu 18, it will be compiled with too old Python. Therefore, I am afraid you are out of luck. KiKit from the beginning targets Python 3.7 and I do not plan to introduce support for older versions of Python. The reason why you can install version 0.6 is that until this version, I did not strictly enforce the version of the interpreter.

The solution is to either upgrade (e.g., to the latest LTS) or to compile KiCAD by yourself to support newer Python. If you don't mind the absence of GUI, you can run KiKit inside Docker.

Unfortunately, Python and KiCAD are not friendly in this area and there's not much I can do.

closed 2022-01-10 by yaqwsx

sslupsky, 2022-01-14:
@yaqwsx I don't think this is the problem @MarkJeronimus is having.  I think I am observing the same issue he is but I am running on macOS.  When I run kikit from the command line, I can confirm that I have version 1.0.2 installed.  My python version is 3.9.9

```
barbie-en0:~ stevenslupsky$ kikit --version
kikit, version 1.0.2
```

```
barbie-en0:~ stevenslupsky$ python3
Python 3.9.9 (main, Nov 21 2021, 03:23:42) 
[Clang 13.0.0 (clang-1300.0.29.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
```

Also, note when I click the "ok" button, the dialog box remains on the screen.  I expect this dialog should close when "ok" is clicked.

[image]

yaqwsx, 2022-01-15:
@sslupsky Since you are running MacOS, your problem is probably connected with #252

ifohancroft, 2022-01-27:
I have the same problem on Linux.

I have Python version 3.10.1
kikit version 1.0.2+14.g3562a29

When I start pcbnew in KiCAD I get the same error about the backend. The window also can't close via the ok button. I have to press the X a couple of times (also I think there are two windows with the error).

yaqwsx, 2022-01-27:
@ifohancroft When you open Pcbnew and go to Tools -> Scripting Console. What version of Python is there? What distribution do you use?

ifohancroft, 2022-01-27:
@yaqwsx Haha I was actually just checking that. The version of Python there is 3.10.1 as well.
I am using Manjaro.

yaqwsx, 2022-01-27:
When you enter into the Pcbnew Scripting console:
```
import sys
sys.executable
```
what do you get?

And what is the output of `which python` and `which python3`? I suspect that KiKit is installed in a different Python interpreter than Pcbnew uses, therefore it cannot load KiKit.

ifohancroft, 2022-01-27:
Hmm, I have found a couple of things. Which of those contribute to the problem, I don't know, but here they are:

1. Even though KiCAD uses Python 3.10.1, it has provided the pcbnew Python module for Python 3.9. Probably the reason KiKit can't find it. (Shouldn't it still be able to find it when I set the env in the terminal pointing to the folder containing it and then run kicad in the same terminal?)

2. The reason the button wasn't closing the window for me, was line 28 in the PCM plugin, True was written as true. Even though I have just compiled kikit from an AUR git package which pulls the source from git, I still had an old code for kikit in KiCAD for some reason (I have just realized that probably the plugin in KiCAD's addon manager is old).

3. I presume you want me to execute `which python` and `which python3` in my OS's terminal, not in Pcbnew's Scripting console? I get /usr/bin/python and /usr/bin/python3. Then I executed `file /usr/bin/python` and it told me python is a symbolic link to python3. I checked python3 with `file /usr/bin/python3` and sure enough it's the correct version. It's a symbolic link to python3.10. I am guessing the problem lies in the pcbnew python module KiCAD provides (as I mentioned in 1.).

4. I can't execute or write anything in Pcbnew's scripting console. Every time I open it, I also get a window showing stderr and it's full of errors. Again, I think it's related to KiCAD providing Pcbnew's Python module for a different Python version than the one it uses.

P.S. Who is responsible for KiCAD's plugins from the plugin manager? If it's them - notify them they are distributing an older version of kikit. If it's you - please update it as your repo includes fixes not yet available in the plugin obtainable via KiCAD's plugin manager.

P.P.S. I guess I need to find Manjaro's bug tracker and report that it provides its module for an older version. As well as perhaps compile my own KiCAD for the time being.

Thank you for guiding me through all of this.

Oh! And before I forget. The installation guide: https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md is missing some steps. More specifically, check the second to last sentence in https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#installation-on-linux-and-macos. It mentions registering the plugins and enabling the libraries manually and those two link to:

https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#enabling-plugins

and

https://github.com/yaqwsx/KiKit/blob/master/doc/installation.md#enabling-kikit-annotation-footprint-library

but those two sections are missing.

P.S. Is the footprint library necessary or can I use kikit with KiCAD but just installing kikit, and then the kikit PCM plugin? Where/when is the library actually needed/used?

MarkJeronimus, 2022-01-30:
I've always had the same thing with the scripting console. When opened I get a stack trace ending in `AssertionError: No wx.App created yet` and a crash popup from Apport (Ubuntu's crash reporter) with the same message.

yaqwsx, 2022-01-30:
This seems like a problem in KiCAD; I recently notices some of the stuff regarding plugins is broken.

Regarding the "registering plugin" - I removed the links. They refer to a no longer supported method of installing the GUI plugin. PCM is the only supported version.

The version in the PCM is indeed 1.0.0. The only thing that changed in the GUI plugin is the fixed bug with the broken OK button. I was considering it as minor back then and I was hoping to release a 1.0.3 version in a few days (that didn't happen) so I postpone the update procedure in PCM. As it is a manual process that requires third parties involved.

The library is only necessary if you want to use the annotation symbols. There is no other use for it yet.

MarkJeronimus, 2022-01-30:
I just incidentally stumbled across a different plugin that won't work, and when I go to Action Plugins->⚠️, I see in the stack trace that KiCad is using the Python3.6 executable, even though I've disabled version.

```
$ sudo update-alternatives --config python3
There are 2 choices for the alternative python3 (providing /usr/bin/python3).

  Selection    Path                       Priority   Status
------------------------------------------------------------
  0            /usr/local/bin/python3.10   2         auto mode
  1            /usr/bin/python3.6          1         manual mode
* 2            /usr/local/bin/python3.10   2         manual mode

Press <enter> to keep the current choice[*], or type selection number: ^C

$ file /usr/bin/python3
/usr/bin/python3: symbolic link to /etc/alternatives/python3

$ file /etc/alternatives/python3
/etc/alternatives/python3: symbolic link to /usr/local/bin/python3.10

$ file /usr/bin/python3.6
/usr/bin/python3.6: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=a7c130aee86ad964af03bfe6178ce8b23417d8fc, stripped
```

The only reference left on my system to python3.6 is the actual executable, which I cannot apt-remove without destroying my system. I cannot find for the life of me how KiCad even decides which executable to use, but if it's hardcoded to `/usr/bin/python3.6` then that's a really odd choice (hence I don't expect it to be that way)

ifohancroft, 2022-01-30:
Apparently the problem wasn't my distro's KiCad build. I compiled it on my own now and am still experiencing the same problem.

Since the KiCad console keeps giving me errors though, I opened a bug report: https://gitlab.com/kicad/code/kicad/-/issues/10679