yaqwsx/KiKit #285: NextPCB is missing a layer in generated gerbers with some profiles

krikun98, 2022-01-08:
[Nextpcb ](https://www.nextpcb.com/pcb-quote) is not detecting the top copper layer 2-layer board gerbers generated with `kikit fab jlcpcb` or with the `oshpark` profile.
Gerbers generated with `kikit export gerber pcb/pinky/nijuni.kicad_pcb gerbers/test` have only one layer detected too.

It works properly when using the `pcbway` profile.

Manually generated gerbers with this version of Kicad and the following settings work fine too: 
```
Application: KiCad PCB Editor (64-bit)

Version: (6.0.0), release build

Libraries:
	wxWidgets 3.1.5
	libcurl/7.78.0-DEV Schannel zlib/1.2.11

Platform: Windows 10 (build 19044), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Dec 24 2021 19:17:02
	wxWidgets: 3.1.5 (wchar_t,wx containers)
	Boost: 1.76.0
	OCC: 7.5.0
	Curl: 7.78.0-DEV
	ngspice: 35
	Compiler: Visual C++ 1929 without C++ ABI

Build settings:
	KICAD_USE_OCC=ON
	KICAD_SPICE=ON
```
[image]

# To Reproduce

Source board: https://github.com/krikun98/jorian840/blob/holyiot/pcb/solder/jorian.kicad_pcb

KiKit command: `docker run -v=$(pwd):/kikit -w=/kikit --rm yaqwsx/kikit:nightly kikit fab jlcpcb --no-assembly "$file" gerbers/"$name"_"$short_option" --no-drc`

I'm building all the gerber files in GH Actions with [this build script](https://github.com/krikun98/jorian840/blob/holyiot/build.sh).

`jlcpcb` profile gerbers:
[pcb_solder_gerbers.zip](https://github.com/yaqwsx/KiKit/files/7834194/pcb_solder_gerbers.zip)

`pcbway` profile gerbers (work properly):
[pcb_solder_gerbers.zip](https://github.com/yaqwsx/KiKit/files/7834196/pcb_solder_gerbers.zip)

`oshpark` profile gerbers:
[pcb_solder_gerbers.zip](https://github.com/yaqwsx/KiKit/files/7834203/pcb_solder_gerbers.zip)

`kikit export` gerbers:
[export_gerbers.zip](https://github.com/yaqwsx/KiKit/files/7834208/export_gerbers.zip)

Manually generated gerbers (work properly):
[gerbers.zip](https://github.com/yaqwsx/KiKit/files/7834186/gerbers.zip)

# Screenshots

Nextpcb with manually generated gerbers (Fine):
[image]

With `kikit fab jlcpcb`:
[image]
The mask/silk layers are picked up, but not the copper.

# Version

Specify which version of KiKit do you use:
- the output of `kikit --version`: `kikit, version 1.0.2+3.gb30ccbd`
- source of installation (PyPI or upstream version): Nightly docker image
- your operating system: GH Actions ubuntu-latest
-

krikun98, 2022-01-09:
I also messaged Nextpcb support in case this is an issue with their viewer.

krikun98, 2022-01-11:
Nextpcb support has forwarded the issue to their software team.

yaqwsx, 2022-01-11:
It is strange that they can process the backside, but cannot process the front side. Nevertheless, the only difference between JLC and PCBWAY settings is if a minimal Gerber header should be used.

This is more a feature request; I am happy to add NextPCB but first, I would like to let them resolve the issue that they cannot load the front side even when it is the same format.

krikun98, 2022-01-11:
I think the files are named differently in the different archives? A feature request would be to add a manufacturing tolerances preset or something, this seems more like a bug.

yaqwsx, 2022-01-11:
Where's the bug? You are trying to send files prepared for JLC PCB/OSHPark/PcbWay to NextPCB. Support for Next PCB is a new feature from this point of view (not a complicate one, but still a feature).

krikun98, 2022-01-11:
Well, I thought gerbers were an industry standard format, and not specifically prepared for each fab. I'm not talking about manufacturing tolerances here, just basic data that's missing. PCBway's format works fine, and I'm not aware I'm preparing gerbers for any fab when I'm generating them manually, yet I'm still expecting them to at least display correctly everywhere.
NextPCB's viewer does not detect a layer file that's in the archive, and their support thinks it's an issue. I should've been more clear: it's a bug with their viewer, not `kikit`.

closed 2022-05-03 by yaqwsx