yaqwsx/KiKit #286: Settings on panel file are not saved when re-run.

troisiemetype, 2022-01-12:
Hello !

Let' start by a big thank you for KiKit ! A so well design tool that I could make my first panelization ever in just a few hour, including learning the tool !

The problem I encounter is not really a bug, more something that could be improve.
When creating a panel from the command line, the result file is created the first time, then replaced each time the CLI is re-run. It has the drawback of resetting all that has been set in this file.

Example : say I've created a panel from a board intended to be a prototype. When plotting gerbers, I check some options to use auxiliary axis as origin, do not tend via, and force plotting of invisible values and refs. Then I find that a modification has to be made to the bare pcb, or the panel, so the CLI is run again. All the settings made previously have to be made again. I would be great if once the file has been created, the settings are saved when script is re-run. (Maybe copied, buffered, and repatched on the new file ?)

Step to reproduce : create any panel with CLI, then change some settings in the create kicad_pcb file, then save, then run the script again.

```
kikit panelize -p json_fil.json input.kicad_pcb output.kicad_pcb
```

Specify which version of KiKit do you use:
- version 1.0.2
- Installed via PyPi
- Running on linux, Lubuntu 20.04

yaqwsx, 2022-01-18:
Thank you for your input. Currently, I am not sure if I like the proposal or not. The KiKit design philosophy is to "automate processed", thus we do not assume anywhere that the files are modified by hand at any moment. The goal is to have a reproducible workflow, that only depends on the inputs, and every time you run it, you get the same results. This is why KiKit deliberately rewrites the panelized board.

When you need to change anything about the output, it should be changed via a script in the "optimal KiKit workflow". Similarly, when you need to export anything from the board all settings should be located in a script. This after all why `kikit fab` exists. Since the export settings are not driven by the board design, but by the fact which manufacturer you will use. From this point, having export settings baked in the board actually makes no sense.

So if we implement something like "preserve board settings", I would only accept it if it was triggered by a switch that is by default off.

troisiemetype, 2022-01-19:
Hello,

Thank you for this answer. You are absolutely right, the all-automated way is the best way to go. For the two designs I've just sent to fab, I've used only the panelization tools, but did all the exports by hand, because that's what I'm used to. I'll be sure to use the fab function instead next time. And take some time to understand how to writes scripts for special cases.

Note : another drawback of the rewritten file is the output folder changes as well. But once again, if I use the CLI there is no more problem.

Maybe having it as an option could be a good thing, but reading you I'm not sure I can see a case where it would.

Again, thank you.

yaqwsx, 2022-05-03:
Since KiKit now supports the placement of board the board in the page and auxiliary origin and the rest (layer visibility) is only graphical, I decided to not implement this feature as it is against the philosophy of reproducible workflow.

What we should do instead is to either provide CLI options or add a scripting API to set the visibility of layers so the user can specify which layers should be visible in the final board.

troisiemetype, 2022-05-04:
Hi Yaqwsx.

Since that "issue" I've used Kikit more, and wrote some python script to panelize boards that could not be made with the CLI. It's obviously, as you said, the way to go when specific things are needed. Not the easiest way, because the documentation of the Kicad API is hard to find (that is : I couldn't find it). But clearly the way to go, as this is as errors are avoided, as you first stated.