yaqwsx/KiKit #288: Error No such file or directory

MiroslavBurda, 2022-01-14:
**Describe the bug**: In KiCad command prompt after command written below was written: An error occurred: [Errno 2] No such file or directory: 'panel.kicad_pro'
No output files produced 

**Expected behavior**: Kikit should panelize board according commands in file Robotarna555.json.

# To Reproduce

Source board: 
[Robotarna555.zip](https://github.com/yaqwsx/KiKit/files/7872320/Robotarna555.zip)

KiKit command: kikit panelize -p Robotarna555.json Robotarna
555.kicad_pcb panel.kicad_pcb
[Robotarna555 (2).zip](https://github.com/yaqwsx/KiKit/files/7872326/Robotarna555.2.zip)

# Version

Specify which version of KiKit do you use: 
- the output of `kikit --version`  1.0.2
- source of installation (PyPI or upstream version) ?? you know :-)
- your operating system win 8.1

yaqwsx, 2022-01-20:
The problem happens when the project of the **source** board is opened in KiCAD. When you close, it works. I wasn't able to trace down why does this happen yet.

yaqwsx, 2022-01-20:
Related KiCAD issue: https://gitlab.com/kicad/code/kicad/-/issues/10540

yaqwsx, 2022-03-07:
The KiCAD issue was resolved. It will be part of 6.0.3 release.

yaqwsx, 2022-03-17:
KiCAD 6.0.3 was released and DRC checking works.

closed 2022-03-17 by yaqwsx