yaqwsx/KiKit #289: Panelize does not copy title block from original board

dkonigsberg, 2022-01-15:
I'm using KiCad v6 with KiKit v1.0.2

When using `kikit panelize` to create a panel of a board, the panel's `kicad_pcb` file does not copy over any of the page settings from the original PCB design. Of more specific concern, it does not copy over any of the text in the "title block" section of the page settings.

This is an issue because KiCad v6 now allows you to reference the title block fields in silkscreen text, and that feature is broken if those fields are not carried over.

I've started using that feature to avoid having to manually synchronize a PCB's revision and date between multiple places.

yaqwsx, 2022-01-16:
This was addressed in 9d0a4cf

closed 2022-01-16 by yaqwsx

esden, 2022-03-11:
Sorry for it taking so long. I tested with the new release and it is indeed fixed. Thank you! :)